JMRI/JMRI #8676: Online Jacoco listing removed class

icklesteve, 2020-06-12:
In a number of places, I've spotted that the online Jacoco coverage reports are listing removed classes with original details ( line count, instruction, complexity etc. ).

All is fine when I'm running coverage reports locally.
Not a major bug, just makes it awkward to compare what's there at present with what I'm developing locally.

One example is https://builds.jmri.org/jenkins/job/development/job/jacoco/lastBuild/jacoco/jmri.jmrix.can.cbus.swing.console/
jmrix.can.cbus.swing.console.CbusConsoleSendPane was added in #8160 
and removed from the package in #8349 when it was moved up a level to jmrix.can.cbus.swing.CbusSendEventPane.

icklesteve, 2021-03-30:
Still a bug, the listings are getting more out of date.

DanielBoudreau, 2021-09-28:
The 3 files are still there.

https://builds.jmri.org/jenkins/job/development/job/jacoco/347/jacoco/jmri.jmrit.operations.setup/

bobjacobsen, 2021-09-28:
The `AllTest` task does the sampling, which is then analyzed by the `jacoco` task.  I cleared `AllTest`, but it's consistently failing so hasn't yet recreated the sampling info.

bobjacobsen, 2021-09-28:
Could you check Jacoco run 349 please?
https://builds.jmri.org/jenkins/job/development/job/jacoco/lastBuild/

DanielBoudreau, 2021-09-28:
All good. The three old deleted files are now gone from the JaCoCo coverage report.

Dan

.

DanielBoudreau, 2021-09-28:
Also good in jacoco_2.  Old files not shown in report.

https://builds.jmri.org/jenkins/job/development/job/jacoco_2/2/jacoco/jmri.jmrit.operations.setup/

bobjacobsen, 2021-09-28:
Thanks for checking!

Jacoco_2 was my trying to fix this by making a completely new task.  That’s what led me to `AllTest` as the out-of-date place.  Will remove shortly.

closed 2022-04-21 by bobjacobsen