JMRI/JMRI #8680: JMRI SPROG Throttle broken by #8226

crosland, 2020-06-13:
@trainman419 Please see the report in the users group https://groups.io/g/jmriusers/topic/74802364?p=Created,,,20,2,0,0::recentpostdate/sticky,,,20,2,0,74802364

> In normal throttle mode the speed slider window shows a text drop-down with "128" displayed.
> The speed scale shows -50% to -100%. On clicking the slider or turning the scroll wheel on the mouse the speed jumps to 100% and the locomotive goes to full speed.
> Re-clicking the slider the speed jumps to -50% and the locomotive stops.
> 
> Clicking the text drop-down and selecting "28" the speed scale shows 0% to 100% and the slider works normally.
> Re-selecting "128" the slider still works normally.

@Lionel642 Please ensure your fixes are pushed to 4.19.8 so that they make it onto 4.20

rhwood, 2020-06-13:
See #8226 (issues do not link from titles)

crosland, 2020-06-13:
PS if you need any SPROG specific help, or testing, let me know.

Lionel642, 2020-06-13:
Just pushed to the branch, should be fixed.

crosland, 2020-06-13:
Which branch did you push to? It's still broken on master.

<author@noreply.github.com>
Sent: 13/06/2020 15:51:04

rhwood, 2020-06-13:
The PR containing the push is #8681

crosland, 2020-06-13:
Doh! Sorry, I moved too quickly, it isn't merged yet :)

crosland, 2020-06-13:
I cloned https://github.com/Lionel642/JMRI/tree/release-4.19.8 to my local machine, but the SPROG throttle is still broken.

I've clearly done something wrong as most of the changes do not appear in my local copy.

How do I get a copy of the new code?

Lionel642, 2020-06-13:
Strange, I can see all my changes in :
https://github.com/Lionel642/JMRI/blob/release-4.19.8/java/src/jmri/jmrit/throttle/ControlPanel.java

crosland, 2020-06-13:
I Manually pasted the new file into my local copy and it works. Thank you!

trainman419, 2020-06-13:
Sorry about that. Let me take a look.

trainman419, 2020-06-13:
I had a look through #8681 and reviewed the fix there; looks good! Thanks for the quick fix @Lionel642

crosland, 2020-06-23:
Confirmed fix in 4.19.8

closed 2020-06-23 by crosland