JMRI/JMRI #8682: Automatic Trains threads now running on AWT thread causing system hangs and accidents.

Pugwash1, 2020-06-13:
in #8412, in dispatcher "new Thread(" was replaced with  "jmri.util.ThreadingUtil.newThread(" in a number of places, which doesnt immediatly appear to be a problem. But last night we had an accident that caused expensive damage to a TGV set when JMRI locked up. It was discoverd that there were long running threads running on the AWT thread, effectively bring the entire system to a halt. It may be caused by #8479. 
This needs to be fixed before the next production release.

rhwood, 2020-06-13:
I seriously doubt #8479 caused your issue since that PR moved activity *off* the EDT / Swing / AWT thread.

You need to provide a way to reproduce, logs, and thread dumps for us to be able to address this in any production or test release.

Pugwash1, 2020-06-14:
The obvious answer is usually correct. The routine is called directly from a button, even though it was never designed to be used like that, and the odds of the sleep ever being called in that curcumstance are quite long.

bobjacobsen, 2020-06-16:
Is this resolved?

rhwood, 2020-06-16:
I can’t tell that any work has been done to investigate or resolve this.

Pugwash1, 2020-06-16:
Im working on it. I know exactly what the problem is: AutoTrainsFRame, line approx 438
                    aat.getAutoEngineer().setHalt(true);
                    aat.setTargetSpeed(0.0f);
                    aat.waitUntilStopped();
                    aat.getAutoEngineer().setHalt(false);
The waituntilstopped() should always be called on its own thread. The settargetspeed is supufluous, the sethalt(true) stops the train and it ignores everything until the sethalt(fals).. Depending on parameters it can take up to a second or 2 to react to the halt, a single train using defaults would react within 100ms.

Pugwash1, 2021-03-11:
Fixed

closed 2021-05-10 by stale[bot]