JMRI/JMRI #8709: "Add New Connection" tab disappears if not completed

mstevetodd, 2020-06-18:
In Edit, Preferences, Connections there is an "Add New Connection" tab (Plus symbol). This tab is lost if you add a new connection, select a Manufacturer, then click the "X" to delete the new connection without Saving. Seen on 4.19.8 on Linux and today's master on Win10. 
See screenshot sequence below:
[image]
[image]
[image]

rhwood, 2020-06-18:
See also #8679

silverailscolo, 2021-01-04:
Fix is in PR #9271

closed 2021-01-17 by silverailscolo