JMRI/JMRI #8730: Enhancement request: VSDecoder file location is static and not configurable like script location

mrchuk, 2020-06-25:
Like many people, I'm slowly accumulating a library of custom VSD files. Every time there's an upgrade these need to be physically copied or moved to resources/vsdecoder. While this is no great chore I suppose, it would seem to be logical to make the location of VSD resources configurable in the same way that script location is in PanelPro/Edit/Preferences. This would make upgrades a little more seamless. 
I don't know offhand whether there are likely to be any backwards compatibility issues with future versions of VSD, but this could be covered in Release Notes if it came up.

bobjacobsen, 2020-06-25:
(N.B:  I'm not a user of the VSD software, so please forgive mistakes)

A quick look though the code found two places that seem to reference a /vsdecoder/ file. First is finding 

```
java/src/jmri/jmrit/vsdecoder/VSDecoderPreferences.java:        _defaultVSDFilePath = FileUtil.getExternalFilename("program:resources/vsdecoder");
```

```
java/src/jmri/jmrit/vsdecoder/VSDecoderManager.java:        String dirname = FileUtil.getUserFilesPath() + "vsdecoder" + File.separator; // NOI18N
```

It seems that a "default" path is set up, then stored to a VSD preferences file and reloaded later.  Could you check if you have a file named VSDecoderPreferences.xml in your JMRI preferences directory?

There's also a VSD preferences pane, but I don't see where it's made available in the preferences. Do you know how to get to it?

Thanks

bobjacobsen, 2020-06-25:
I think I found the preferences:   Main Window -> Debug Menu -> Virtual Sound Decoder -> VS Decoder Manager -> (new Window) -> Edit -> VSDecoder Preferences

[image]

Maybe we should also have that show with the other preferences?

mrchuk, 2020-06-25:
Obviously I didn't know this was there.
This will do for me. Enhancement request withdrawn unless someone wants to run with it.

closed 2020-06-25 by mrchuk