JMRI/JMRI #8742: User issue:  Set State of Turnouts in Existing or New Route - JMRI.4.19.8+Rbab3d0f33

dsand47, 2020-06-28:
@pabender 

The state selection combo for output turnouts is based on the turnout state for the connection in `AbstractRouteAddEditFrame`:
```
InstanceManager.turnoutManagerInstance().getClosedText()
```
The state selection is based on `Bundle.getMessage("TurnoutStateClosed")` in `RouteTurnout`.

For XpressNet turnouts `Set Close (-)` does not match `Set Close`.

The output sensor section does not use the connection specific state names.

pabender, 2020-06-28:
@dsand47 this appears to have been an issue even before the recent refactoring of the table code.  There is definitely a mis-match of state texts in routes.

I’ll have to look at where this moved, but here is an example from before the refactoring:

https://github.com/JMRI/JMRI/blob/582bd498531f90abe23a8591e787e4542e12773d/java/src/jmri/jmrit/beantable/RouteTableAction.java#L94-L109

dsand47, 2020-06-28:
I have not looked at the old code.  My guess is that everything was in one class so that the connection based turnout state strings were used for both creating the combo box and comparing the selection.

The easy fix is to remove the two lines that replace the standard bundle based state strings with the connection based state strings.  Or change RouteTurnout to use connection based state strings.

pabender, 2020-06-28:
> I have not looked at the old code.  My guess is that everything was in one class so that the connection based turnout state strings were used for both creating the combo box and comparing the selection.
> 

As I showed in the code snippet I linked to, the combo box was created with some options using the string from the TurnoutManager defined names and others coming from the bundle.  They really ALL should come from the TurnoutManager defined names... because the manufacturers don’t all use the same names, and the manager defines our link to the system specific terminology.

If we expect users to be able to use hand held controllers to trigger a route, and I do, JMRI’s labels should be consistent with those controllers.

> The easy fix is to remove the two lines that replace the standard bundle based state strings with the connection based state strings.  Or change RouteTurnout to use connection based state strings.

But I don’t think that is the correct fix.

This is looking at the state of a turnout.  The set of states for the turnout are limited to those defined in jmri.Turnout.  No other states exist.

The human readable names are mapped to the states.  Those human readable names should NOT impact how panel files load and operate.

Now, if I am going to fix this, I need a test that shows me how this breaks.  A user file is fine.

dsand47, 2020-06-29:
Here is a sample xml file.  Open it with a profile using XpressNet.  Go to the Route table, edit the turnouts route and select the state.

[turnouts.xml.txt](https://github.com/JMRI/JMRI/files/4843206/turnouts.xml.txt)

Here is a screenshot.

[image]

Notice that the XpressNet state strings are even being used for internal turnouts.

The problem is that the other classes are looking for `Set Closed` not `Set Closed (-)`.

pabender, 2020-06-29:
> Here is a sample xml file.  Open it with a profile using XpressNet.  Go to the Route table, edit the turnouts route and select the state.

Ok, I will take a look.

> Notice that the XpressNet state strings are even being used for internal turnouts.

That is a different issue than that the system specific text is being used.  This is because the state names are requested only once, and the proxy manager is being asked for what to call the values.

It is unclear what the proxy manager should report in this case.  The current behavior appears to be that it just picks one on the list to ask.  In reality, the choice should be dynamic per row of the table, I.e. we should ask the manager for the specific turnout.

But, this would have been an issue before any refactoring occurred as well, the choice of values might have been different however.

In other words, this is a subtle bug that has existed for a while.

dsand47, 2020-06-30:
> 
> But, this would have been an issue before any refactoring occurred as well, the choice of values might have been different however.
> 
> In other words, this is a subtle bug that has existed for a while.

Before refactoring, the combo box content creation **and** the selection processing were in `RouteTableAction`, which used connection specific strings.  This was not an issue since both processes used the same strings.

After refactoring, `AbstractRouteAddEditFrame` uses the connection specific name to create the combo box state strings while `RouteTurnout` uses the bundle derived turnout state strings.  This fails for XpressNet (and possibly other) connections.

I realize this repeats what I said earlier, but I feel this needs to be fixed before 4.20.

bobjacobsen, 2020-06-30:
> It is unclear what the proxy manager should report in this case. The current behavior appears to be that it just picks one manager on the list to ask. In reality, the choice should be dynamic per row of the table, I.e. we should ask the manager for the specific turnout.

Right now, it uses the default `TurnoutManager`, because it is providing (in effect, defining) the default terminology.

Individual Turnouts don't currently provide a direct way to access the terminology, but of course they easily could. I think only Lenz (XNetTurnoutManager) and DCC++ use this facility.  

If somebody does decide to add this, please have "get terminology" return something like a reference to an effectively final array so that it can be reused in the all the rows of the table.  The turnout table on some large layouts uses an impressively large amount of memory, and everything we can do to push that down would be a good thing.

Also, while we're digging into the code for rendering, note the arrow below.  Is that rendererManpSensor1 an error? Should it be rendererMapSensor2?

[image]

pabender, 2020-07-01:
@dsand47 

The old code did this for the update of the text:

https://github.com/JMRI/JMRI/blob/582bd498531f90abe23a8591e787e4542e12773d/java/src/jmri/jmrit/beantable/RouteTableAction.java#L94-L197

That is still getting the text from (as Bob pointed out) the default TurnoutManager, which may be the first one entered.

The new code performs exactly the same operation:
https://github.com/JMRI/JMRI/blob/83d1a5329425ba8d43fb0e57d30f4912b947a6ec/java/src/jmri/jmrit/beantable/routetable/AbstractRouteAddEditFrame.java#L128-L131

The only difference being that the RouteTurnout is using a different constant value.

I've got a change I'll push up soon that will allow the RouteTurnout to use the same constant that the Add and Edit frames use, which gets modified with the code above.

pabender, 2020-07-01:
> Individual Turnouts don't currently provide a direct way to access the terminology, but of course they easily could. I think only Lenz (XNetTurnoutManager) and DCC++ use this facility.

Right.  This is on the manager because it applies to all turnounts managed by the manager.  So the system specific text shows when you are on the turnout table for the specific system.

dsand47, 2020-07-01:
I have a temporary fix for this problem:  Add a simulated LocoNet connection as the first connection.  Have the XpressNet connection second and set the defaults to the XpressNet connection.

Since `InstanceManager.turnoutManagerInstance().getClosedText()` uses the default (first) connection it now returns Closed and Thrown without the + and -.

pabender, 2020-07-01:
> I have a temporary fix for this problem: Add a simulated LocoNet connection as the first connection. Have the XpressNet connection second and set the defaults to the XpressNet connection.
>
> Since [...]

PR #8760 makes the Add/Edit frame table behavior be consistent with the TurnoutTable's All tab, which uses the test from the Default Turnout Manager (XPressNet in the case we are discussing).

dsand47, 2020-07-06:
Fixed at 4.19.9.

closed 2020-07-06 by dsand47