JMRI/JMRI #8751: RosterEntry icon not displayed on web layout editor panel

andybadshot, 2020-06-29:
When a RosterEntry is set as the value in a Memory variable, the default name is on the web rather than the image e.g. "RosterEntry: M50926 AW_Class_108.xml    AW Class 108 10  LokSound Micro V4.0 ESU LokSound V4.0" -  

It shows correctly on Layout Editor native panel.

See [https://groups.io/g/jmriusers/topic/rosterentry_icon_not/75188171](https://groups.io/g/jmriusers/topic/rosterentry_icon_not/75188171) for details

Andy Brown

andybadshot, 2020-06-29:
[WebItemFaults.jmri.zip](https://github.com/JMRI/JMRI/files/4846117/WebItemFaults.jmri.zip)

mstevetodd, 2020-06-29:
Andy, you had both a memoryicon and a blockcontentsicon on the panel, but only one was being set by the jython. I added the other to the script, since I needed to test both with my change. Attaching new version. 
[RosterEntryToMemory.py.txt](https://github.com/JMRI/JMRI/files/4848291/RosterEntryToMemory.py.txt)

mstevetodd, 2020-06-29:
This should be fixed by #8754 going thru CI now.

PR #8754 (2020-06-29): client-side panels: support rosterEntries in blocks and memories
Fixes #8751.
closed 2020-06-29 by mstevetodd