JMRI/JMRI #8766: Enhancement request: Add logix for Blocks

mrchuk, 2020-07-02:
Currently JMRI supports logix for OBlocks in both state variables and Actions. Logix lacks support for Blocks in either category. This appears to be an omission necessitating workarounds by affected users.

Discussion of this is occurring in this jmriusers thread: https://groups.io/g/jmriusers/topic/regarding_logix_why_no_logix/75252478?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,75252478

dsand47, 2020-07-02:
I have started working on this request.

The first question concerns which variables and actions should be implemented.  Here are my current ideas:

- Variables
  - Block Occupied ( True | False )
  - Block Allocated ( True | False ) -- Related LayoutBock useExtraColor
  - Block Value Equals ( String | Memory Variable Content )
- Actions
  - Set Block Occupied ( True | False )
  - Set Block Allocated ( True | False ) --Related LayoutBock useExtraColor
  - Set Block Value using ( String | Copy content from Memory Variable )
  - Copy Block Value content to Memory Variable

pabender, 2020-07-02:
Sent from my iPad

> On Jul 2, 2020, at 5:12 PM, Dave Sand <notifications@github.com> wrote:
> The first question concerns which variables and actions should be implemented. Here are my current ideas:
> 
> Variables
> Block Occupied ( True | False )
> Block Allocated ( True | False ) -- Related LayoutBock useExtraColor
> Block Value Equals ( String | Memory Variable Content )
The contents of the block is an object.  You should be able to compare this to any object we can reasonably build a UI to display as a choice.

One example here is an IDTag ( including the RailCom and Transponding subtypes ).

A couple of example use cases:
1) Play elephant sounds when the car identified by ID tag as the circus train’s elephant car enters the block.  Stop playing when it exits.
2) Update the location of a train in operations when the engine is identified at a location.

There are others I am sure.

dsand47, 2020-07-02:
> The contents of the block is an object.

Thanks.  I have replaced String with Object.

mrchuk, 2020-07-02:
Dave,
The Actions seem complete. To make the functionality versatile, it should include adding Block as an Action Group for Sensor, as is currently available for OBlocks. I'm not sure whether your proposal would encompass this or not.

But if so, it would allow the proposed Actions on triggering one or more sensors, and permit complex sensor inspection where needed.

dsand47, 2020-07-02:
> Dave,
> The Actions seem complete. To make the functionality versatile, it should include adding Block as an Action Group for Sensor, as is currently available for OBlocks. I'm not sure whether your [...]

Can you clarify what you mean by **Action Group for Sensor**?

mrchuk, 2020-07-03:
Right now, in PanalPro you can open Tools/Tables/Logix/Add, create a logix and start with one or more Sensor state variables.
Screenshot:

[image]

When you go to the Consequent Action/Action Group dropdown, Block isn't available but OBlock is. My idea for this enhancement is that Block should be incliuded here, in the same way OBlock is currently.

dsand47, 2020-07-03:
> When you go to the Consequent Action/Action Group dropdown, Block isn't available but OBlock is. My idea for this enhancement is that Block should be incliuded here, in the same way OBlock is currently.

I assumed that is what you meant but I wanted to make sure.

If you look at my first posting you will see two sets of Block options.  The first set are the new State Variables for **Variable Type => Block**, the second set are the new Actions for **Action Group => Block**.

pnbgit, 2020-07-08:
Should block occupied be True|False|Unknown  ?   It can be useful to check for unknown state sometimes.

dsand47, 2020-07-08:
> Should block occupied be True|False|Unknown ? It can be useful to check for unknown state sometimes.

A block has 5 states:
- UNDETECTED -- No sensor.
- INCONSISTENT -- Sensor is Inconsistent .
- UNKNOWN -- Sensor is Unknown.
- UNOCCUPIED -- Sensor is Inactive.
- OCCUPIED -- Sensor is Active.

Other objects, such as sensors and turnouts, don't have Logix support for inconsistent or unknown.

I see the variable and action support of **occupancy** as being convenience options that can also be done by using the related sensor.  The real value is the access to the **block content** field along with access to the LayoutBlock **useExtraColor** state which is used to indicate allocation.

dsand47, 2021-03-30:
Due to the complexity and risk of adding additional objects to Logix, this project is on hold.

LogixNG (Logix Next Generation) is being added to JMRI starting with 4.23.2.  This uses a modular approach, which will make adding new objects such a Blocks easier with less risk.

dsand47, 2021-05-14:
LogixNG including block support is active in test release 4.23.5.

closed 2021-05-14 by dsand47