JMRI/JMRI #8784: Running of single tests broken

bobjacobsen, 2020-07-04:
`./runtest.sh` can be used to run single test classes from the command line.  Recently, it's gotten much slower.

Current results:
```
% time ./runtest.csh java/src/jmri/InstanceManagerTest.java
Jul 04, 2020 3:30:59 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
log4j:WARN No appenders could be found for logger (com.tngtech.archunit.core.PluginLoader).
log4j:WARN Please initialize the log4j system properly.
log4j:WARN See http://logging.apache.org/log4j/1.2/faq.html#noconfig for more info.
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer

Test run finished after 2208 ms
[         4 containers found      ]
[         0 containers skipped    ]
[         4 containers started    ]
[         0 containers aborted    ]
[         4 containers successful ]
[         0 containers failed     ]
[        22 tests found           ]
[         0 tests skipped         ]
[        22 tests started         ]
[         0 tests aborted         ]
[        22 tests successful      ]
[         0 tests failed          ]

./runtest.csh java/src/jmri/InstanceManagerTest.java  74.91s user 5.85s system 223% cpu 36.099 total
```

InstanceManagerTest runs OK, but very slowly and with a message about logging for `com.tngtech.archunit.core.PluginLoader`

Under 4.91.8:

```
% time ./runtest.csh java/src/jmri/InstanceManagerTest.java
Jul 04, 2020 3:40:27 PM org.junit.platform.launcher.core.LauncherConfigurationParameters fromClasspathResource
INFO: Loading JUnit Platform configuration parameters from classpath resource [file:/Users/jake/Documents/Trains/JMRI/projects/JMRI/target/test-classes/junit-platform.properties].
Using org.netbeans.jemmy.drivers.APIDriverInstaller driver installer

Test run finished after 1821 ms
[         3 containers found      ]
[         0 containers skipped    ]
[         3 containers started    ]
[         0 containers aborted    ]
[         3 containers successful ]
[         0 containers failed     ]
[        22 tests found           ]
[         0 tests skipped         ]
[        22 tests started         ]
[         0 tests aborted         ]
[        22 tests successful      ]
[         0 tests failed          ]

./runtest.csh java/src/jmri/InstanceManagerTest.java  5.14s user 0.95s system 154% cpu 3.945 total
```

It's over a minute faster to start up.

rhwood, 2020-07-04:
I would not be surprised if I got the JUnit 5 launcher wrong or am doing something else incorrectly in jmri.util.junit.TestClassMainMethod

pabender, 2020-07-05:
One thing I noticed since the mass Junit5 conversion is I that the ArchUnit tests are now being run with a JUnit5 Engine because of the way they appear in Intellij.

This means that, if we've told TestClassMainMethod to invoke it, that JUnit is invoking 3 engines (Jupiter,Vintage, and ArchUnit) instead of the 2 it previously invoked.

Maybe it is time we moved away from using our own test class main method in RunTest to using the JUnit5 ConsoleLauncher. We already use this for several of the ant tasks (test-single and test-single-coverage in particular) but it can also be run on the command line.

bobjacobsen, 2020-07-05:
Something in what @pabender, along with the `com.tngtech.archunit.core.PluginLoader` reference in the message (which happens whether ArchitectureTest is run or now) suggested I do a thread dump during the delay.  A lot of time seems to be spend in (some lines removed):
```
	at sun.nio.fs.UnixFileAttributes.get(UnixFileAttributes.java:72)
...
	at java.nio.file.Files.walkFileTree(Files.java:2706)
...
	at com.tngtech.archunit.core.importer.Location$FilePathLocation.getAllFilesBeneath(Location.java:378)
	at com.tngtech.archunit.core.importer.Location$FilePathLocation.iterateEntriesInternal(Location.java:366)
	at com.tngtech.archunit.core.importer.Location$1.call(Location.java:74)
...
	at com.tngtech.archunit.thirdparty.com.google.common.cache.LocalCache$LocalManualCache.get(LocalCache.java:4899)
	at com.tngtech.archunit.core.importer.Location.iterateEntries(Location.java:153)
...
	at com.tngtech.archunit.core.importer.ClassFileImporter.importPackages(ClassFileImporter.java:220)
	at com.tngtech.archunit.junit.ArchUnitTestEngine.getContainedClasses(ArchUnitTestEngine.java:146)
	at com.tngtech.archunit.junit.ArchUnitTestEngine.resolveRequestedPackages(ArchUnitTestEngine.java:102)
	at com.tngtech.archunit.junit.ArchUnitTestEngine.discover(ArchUnitTestEngine.java:81)
	at org.junit.platform.launcher.core.DefaultLauncher.discoverEngineRoot(DefaultLauncher.java:181)
	at org.junit.platform.launcher.core.DefaultLauncher.discoverRoot(DefaultLauncher.java:168)
	at org.junit.platform.launcher.core.DefaultLauncher.discover(DefaultLauncher.java:124)
	at jmri.util.junit.TestClassMainMethod.run(TestClassMainMethod.java:93)
```

So for some (yet unknown) reason, the ArchUnit code is being fired to process all the class files, even if the ArchitectureTest file isn't being referenced.

Interestingly, this happens even if the target/test-classes/jmri/ArchitectureTest.class and target/test-classes/jmri/ArchitectureCheck.class files are removed. 

Removing the lib/archunit* files from the class path gets back the original speed, and the extra logging message is gone.  So it seems to be something that happens when those are loaded and somehow invoked by JUnit5.  

I don't think this can be the intended behavior of those files; it's not what we saw under JUnit4, and you usually specify the target files to ArchUnit.

Anybody know what org.junit.platform.launcher.core.DefaultLauncher.discoverRoot or discoverEngineRoot is intended to do?

I found the source for this, but we must be using a different version because discoverRoot is missing and the line numbers aren't even close to lining up.

rhwood, 2020-07-05:
ArchUnit rearchitected their launchers between the Junit4 and Junit5 versions.

Randall

bobjacobsen, 2020-07-05:
Here's the relevant code in `jmri.util.junit.TestClassMainMethod`:

```
    /**
     * Run tests with a compile-selected RunListener.
     * 
     * @param testClass the class containing tests to run
     */
    public static void run(String testClass) {
        SummaryGeneratingListener listener = new jmri.util.junit.PrintingTestListener(System.out); // test-by-test output if enabled
        run(listener, testClass);
        TestExecutionSummary summary = listener.getSummary();
        PrintWriter p = new PrintWriter(System.out);
        summary.printTo(p);
        summary.printFailuresTo(p);
    }

    /**
     * Run tests with a specified RunListener.
     * 
     * @param listener the listener for the tests
     * @param pattern the filter pattern used for test selection
     */
    public static void run(TestExecutionListener listener, String pattern) {
        LauncherDiscoveryRequest request = LauncherDiscoveryRequestBuilder.request()
                .selectors(selectPackage("jmri"))
                .selectors(selectPackage("apps"))
                .filters(includeClassNamePatterns(pattern))
                .build();
        Launcher launcher = LauncherFactory.create();
        TestPlan testPlan = launcher.discover(request);
        launcher.registerTestExecutionListeners(listener);
        launcher.execute(testPlan);
    }
```

The first method is invoked with a `testClass` string like "jmri.InstanceManagerTest' or 'jmri.jmrit.*'. It creates a `TestExecutionListener` and passes it, plus the `testClass` string, to the second method.

That then proceeds to have `LauncherDiscoveryRequestBuilder` builds a LauncherDiscoveryRequest that's set to scan through all the code.  That's passed to the `Launcher`, which invokes an ArchUnit discovery routine that takes a lot of time.

That's always been a bit inefficient (all the code?), but now that ArchUnit has provided a service that looks hard at code during discovery, it's become noticeable. 

I've created PR #8788 to improve this.  It recovers this delay, and puts the speed back to about as fast as it was before the earlier JUnit5 conversion.

bobjacobsen, 2020-07-25:
Fixed by #8788

closed 2020-07-25 by bobjacobsen