JMRI/JMRI #8786: Dispatcher is throwing a NullPointerException in MemoryIcon.flipIcon when changing direction of an automatic train

andybadshot, 2020-07-05:
NullPointerException is thrown when using the Dispatcher with auto trains on a CBUS system when changing direction, as described in https://groups.io/g/jmriusers/topic/75275563 

Proposed change is to add null protection around _namedIcon:
```
    protected void flipIcon(int flip) {
        if (_namedIcon != null) {
            _namedIcon.flip(flip, this);
        }
        updateSize();
        repaint();
    }
```
I'm still trying to find a simple way to reliably reproduce the error to help with testing.
Andy Brown

andybadshot, 2020-07-06:
Raised PR https://github.com/JMRI/JMRI/pull/8801 to solve this issue.

Andy Brown

bobjacobsen, 2020-07-06:
Closed by #8801

closed 2020-07-06 by bobjacobsen