JMRI/JMRI #8799: Dispatcher is throwing a NullPointerException when terminating an automatic train

andybadshot, 2020-07-06:
As described in https://groups.io/g/jmriusers/topic/75266552 there is a NullPointerException thrown sometimes when a Dispatcher auto train terminates. This happens on a Merg Cbus system. 

There appear to be several 'competing' threads calling the jmri.jmrix.AbstractThrottle.finishRecord() method. On occasions 2 of these clash meaning that sometimes an initial test (re == null) is not adequate protection as one thread is setting re = null whilst another thread is processing the method.

Steve Young suggested the simple fix of making the method syncronized:

```
    synchronized protected void finishRecord() {
        if (re == null) {
            return;
        }
        stopClock();
        ....

```

Ad-hoc testing on my test layout seems to indicate that this fix works, with no adverse effect seen. It has only be tried with Merg Cbus hardware - none other is available to me.

Andy Brown

andybadshot, 2020-07-06:
Raised pull request https://github.com/JMRI/JMRI/pull/8800 to solve this issue

Andy

bobjacobsen, 2020-07-06:
Could you do an experiment for me please?

Add the line
```
  log.warn("Debugging", new Exception("traceback"));
```

right at the start of finishRecord, then run your sample cases where you've seen this.

The Cbus throttle really should only be calling this once and on the layout thread.  I'd like to know why it's being called more than one, and particularly which threads that's coming from.  The tracebacks in the log should should that, even if the NPE issue doesn't happen.

Thanks!

Bob

andybadshot, 2020-07-08:
I've tried this (with the 'synchonized' in place) and I only saw one thread calling finishMethod, right up until I had problems with my trains derailing when I manually terminated the Dispatcher - that is the 2nd thread at the end. However, they did not coincide in any way.

Looking back at my original post https://groups.io/g/jmriusers/message/176450, the were 2 different threads. I've copied that stacktrace at the end of this email.

I was doing the investigation in conjunction with Steve Young, who originally suggested a change (mentioned in the original topic) to CbusThrottleManager to (So in jmri.jmrix.can.cbus.CbusThrottleManager it should be safe to remove line 181 ( throttle.throttleDispose(); )

I tried that, but still got a problem, so he suggested the synchonized change. I added that and all seemed well.

I see that Steve has now put in place the throttleDispose fix with https://github.com/JMRI/JMRI/pull/8768

It makes me wonder whether that did solve the problem all along and it was my mistake when testing it. Whatever, I cannot reproduce the problem now based on the latest code (without synchonized)

Closing the issue on the basis that PR https://github.com/JMRI/JMRI/pull/8768 actually solved the problem without requirng the proposed change.

Andy

closed 2020-07-08 by andybadshot