JMRI/JMRI #8808: web panel xml providing should provide flexible path to images

mstevetodd, 2020-07-07:
For background discussion, see https://groups.io/g/jmriusers/topic/75260975
Attaching user's config export to this issue. Note that the user is manually loading file "Ullapool11.xml".
[My JMRI Railroad.zip](https://github.com/JMRI/JMRI/files/4886550/My.JMRI.Railroad.zip)

JMRI panels support "overriding" images by placing new files of the same name in folders of the same name in the resources folder under the "user preferences" folder. These are then used to display the user's icons on panels. The reporting user used this feature and it worked properly on the JMRI panel.

The issue is that when displayed on the client-side web panel, the JMRI images were shown, not the user's override images. The panel xml returns (example):
`/dist/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
according to Randall, this should be 
`/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
which the webserver would use to return the overridden image if it exists on disk, or the JMRI image if not.
I verified this is the case by browsing to 
`http://localhost:12080/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
and the browser displayed the correct overridden image.

This is done when converting from the stored path to the portable path to the web path.
Example: 
When building the icon list for a signalMastIcon, one filename starts as 
`../../../resources/icons/BritishRailSignals/Aspects/PLOff.gif`
which is then converted to
`program:resources/icons/BritishRailSignals/Aspects/PLOff.gif`
which is then converted to
`/dist/resources/icons/BritishRailSignals/Aspects/PLOff.gif`

This last conversion occurs in AbstractPanelServlet.parsePortableURIs(), which calls WebServer.portablePathToURI()
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/web/servlet/panel/AbstractPanelServlet.java#L220
https://github.com/JMRI/JMRI/blob/master/java/src/jmri/web/server/WebServer.java#L171

So, does WebServer.portablePathToURI() need to change to return "/" for "program:" instead of "/dist/"?
Or should the "/dist" be stripped off later?
Or something else?

mstevetodd, 2020-07-07:
@rhwood Is this feature of the webserver documented somewhere? I was (formerly) unaware you could leave off the /dist and the web server would look in /prefs and then /dist.

mstevetodd, 2020-07-07:
I made the change to portablePathToURI() and tested with several panel types and they are working fine. I'm going to create a PR and run the CI tests.

rhwood, 2020-07-07:
> @rhwood Is this feature of the webserver documented somewhere? I was (formerly) unaware you could leave off the /dist and the web server would look in /prefs and then /dist. Does it work just for images, or for any supported file type?

No explicitly. The closest we get is https://www.jmri.org/help/en/html/web/index.shtml#custom

rhwood, 2020-07-07:
> @rhwood Is this feature of the webserver documented somewhere? I was (formerly) unaware you could leave off the /dist and the web server would look in /prefs and then /dist. Does it work just for images, or for any supported file type?

I missed half the question. This works for any file. Besides the link in help, see [user.css](https://github.com/JMRI/JMRI/blob/master/web/css/user.css) and [user.js](https://github.com/JMRI/JMRI/blob/master/web/js/user.js)

rhwood, 2020-07-07:
I have closed #8809 since it does not correctly fix the problem.

> For background discussion, see https://groups.io/g/jmriusers/topic/75260975
> Attaching user's config export to this issue. Note that the user is manually loading file "Ullapool11.xml".
> [My JMRI Railroad.zip](https://github.com/JMRI/JMRI/files/4886550/My.JMRI.Railroad.zip)
> 
> JMRI panels support "overriding" images by placing new files of the same name in folders of the same name in the resources folder under the "user preferences" folder. These are then used to display the user's icons on panels. The reporting user used this feature and it worked properly on the JMRI panel.
> 
> The issue is that when displayed on the client-side web panel, the JMRI images were shown, not the user's override images. The panel xml returns (example):
> `/dist/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> according to Randall, this should be
> `/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> which the webserver would use to return the overridden image if it exists on disk, or the JMRI image if not.
> I verified this is the case by browsing to
> `http://localhost:12080/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> and the browser displayed the correct overridden image.
> 
> This is done when converting from the stored path to the portable path to the web path.
> Example:
> When building the icon list for a signalMastIcon, this filename starts as
> `../../../resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> which is then converted to
> `program:resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> which is then converted to
> `/dist/resources/icons/BritishRailSignals/Aspects/PLOff.gif`
> 
> This last conversion occurs in AbstractPanelServlet.parsePortableURIs(), which calls WebServer.portablePathToURI()
> https://github.com/JMRI/JMRI/blob/master/java/src/jmri/web/servlet/panel/AbstractPanelServlet.java#L220
> https://github.com/JMRI/JMRI/blob/master/java/src/jmri/web/server/WebServer.java#L171
> 
> So, does WebServer.portablePathToURI() need to change to return "/" for "program:" instead of "/dist/"?
> Or should the "/dist" be stripped off later?
> Or something else?

The problem is the returned path (`../../../resources/icons/BritishRailSignals/Aspects/PLOff.gif`) is wrong if it converts to `program:...` and not `preferences:...`. The next conversion (`program...` to `/dist/` is wrong only because bad inputs to `portablePathForURI()` will return bad outputs (i.e. "[garbage in, garbage out](https://en.wikipedia.org/wiki/Garbage_in%2C_garbage_out)")

so something else has to change for the actual path to the icon to be returned from the panel in. I think the error is in this method https://github.com/JMRI/JMRI/blob/5bc67409b76df40742ed36b4548f119eb53ec323/java/src/jmri/web/servlet/panel/AbstractPanelServlet.java#L264-L310
where the assumption is made that all paths are already portable.

mstevetodd, 2020-07-08:
Thanks, Randall.
Can you point me to a routine which does that conversion properly?

rhwood, 2020-07-08:
> Can you point me to a routine which does that conversion properly?

Nothing is doing the conversion improperly. Whatever positionable element that is giving the WebServer a path to an image it did not load (instead of the image it did load) needs to be fixed. That said, you may find that passing the results of `getImageLink()` through `FileUtil.getPortableFilename​(...)` might be a workable bandaid, but again, be aware that you will be attempting to use an input that is indeterminately (in)correct unless the code that generates the XML for that element provides real data.

rhwood, 2020-07-08:
What I'm getting at is there is no problem in how the web server is working; the problem is in some configurexml class or in the class that a configurexml class is generating the XML from.

mstevetodd, 2020-07-21:
closed via #8823

closed 2020-07-21 by mstevetodd