JMRI/JMRI #8812: Construction Lines will not stay hidden when Layout Editor is stored then reloaded

andybadshot, 2020-07-08:
Created to address topic https://groups.io/g/jmriusers/topic/74856474.

Constructions lines for Layout Editor arcs do not always save to xml.

Andy Brown

andybadshot, 2020-07-08:
Create PR https://github.com/JMRI/JMRI/pull/8813 to address this.

geowar1, 2020-07-09:
Ok, found the issue: Arc's don't have construction lines. Circle & Bezier (flex) tracks do. The Arc contextual-popup menu for Arc's should have the show/hide construction lines menu item.

andybadshot, 2020-07-09:
The above may be so, but I can easily recreate the original issue:
-Create new panel
-Add anchor points and joint with Track Segment
-Convert to circle and do not alter it (this leaves the angle as zero
-Hide construction line
-Store
-Inspect xml - no 'hideConLines for track segment
-quit/load panel
-construction line showing

I think I was on the right track with my fix, but it failed the tests as my change moved the hideConLines. I have a mod for that.

andybadshot, 2020-07-09:
Simple panel with circle (zero angle), circle > 0 angle, ellipse and bezier
[ConstructionLines01.zip](https://github.com/JMRI/JMRI/files/4899039/ConstructionLines01.zip)

andybadshot, 2020-07-09:
I've committed a new change that now passes CI tests.

I'm thinking of adding the above xml as a new file to load/loadref directory to create a new test. Is this a good idea? This file confirms the change as it fails before and passes after the change.

rhwood, 2020-07-09:
It is always a good idea to include a test that verifies any code change is behaving as expected.

andybadshot, 2020-07-11:
PR https://github.com/JMRI/JMRI/pull/8813 now merged, so closing issue

closed 2020-07-11 by andybadshot