JMRI/JMRI #8815: Turnout Circle Colour Change

JDWR, 2020-07-09:
I have just updated to 4.20 from 4.18 and I cannot get the Turnout Circle Colour to change to the selected colour in the new version. This worked in 4.18. The Turnout Fill colour option works.

geowar1, 2020-07-09:
I'm not aware of any change to that code since 4.18 but I'll take a look.

geowar1, 2020-07-09:
Everything seems to be working as expected for me… can you provide steps to reproduce?

JDWR, 2020-07-09:
George, JMRI 4.20

Reading about the inability to Hide the Construction Lines for Bezier Curves, I am wondering if this may be the same problem but I do not think so. In 4.18 as soon as I selected to Change the colour of the Turnout Circle, it automatically changed every Turnout circle on the layout without having to do anything else but not on 4.20 and I do store after making such changes. Do we have to do a restart to get the saved changes to work here as well?

Dave

geowar1, 2020-11-28:
can you verify this fixed/broken?

closed 2020-11-28 by geowar1