JMRI/JMRI #8829: Possible Error in Blocks Section of xml file

JDWR, 2020-07-13:
I have been studying the xml code in some detail recently and I think that I may have come across a possible error within the code.
 Where I have an Approach Block that has a directional component to it and also a Turnout attached to it, the line within the description of the element says:
<turnout systemName="LL Entry to Industry" />.
This is the Turnout User Name for the Block, not its systemname.
I also noticed that the element has a space between turnout and systemName and the use of an uppercase N in the name.
Should this read <turnout username="? Are these names just for internal use within the program?

dsand47, 2020-07-15:
Are you talking about the **path** portion of the block xml?  For example:
```
      <path todir="64" fromdir="128" block="IB:AUTO:0006">
        <beansetting setting="4">
          <turnout systemName="XO-1" />
        </beansetting>
      </path>
```
If so, that is normal.  The turnout name in a beansetting can be either the system name or the user name based on the NamedBeanHandle.

JDWR, 2020-07-15:
Thanks for that Dave. 
It just looks odd when reading through the coding to see something labelled SystemName using a User Name.

closed 2021-01-31 by silverailscolo