JMRI/JMRI #8843: Ability to add allowed CORS origins to web server

mmillican, 2020-07-16:
Looking for a way to add allowed CORS origins to the JMRI web server so that I can make an additional web app for JMRI. I've been trying to find this in the HTTP server code, but unable to so far as I'm not super familiar with Java.  I'm happy to make the change if someone could point me int he right direction (if it's possible).

rhwood, 2020-07-16:
That is a non-trivial change to make to JMRI (the web server would need to be modified to add support for configuring a Java EE filter chain).

Can you explain in more details what you trying to do? There may be other ways of do what you want without waiting on that change to JMRI.

Randall

pabender, 2020-07-16:
I'm definitely interested in hearing more about your use case for this.

Can you tell us a little about the architecture of your web app, and how JMRI fits into that architecture?

mmillican, 2020-07-16:
Understood. I was afraid it might be more complicated than I thought.

I'm still in the process of figuring out specific details of what it will do, but since this a blocker to that I wanted to start here. The app would be written in Vue.js, which would likely be hosted on another port, thus requiring CORS to JMRI. I was also thinking that I could write a middleware API that would talk to JMRI and thus not need CORS (the middleware would allow CORS), but that would get tricky with the web sockets.

Right now, some of the ideas I'm thinking of are:
- Ways to manage the roster (ie, see what items are out of service to work on)
- Playing with different ideas for displaying the manifests and conductor views (more closely mirror what's in JMRI?)

@pabender as far as the architecture goes, I'm imagining that this is a standalone app which uses JMRI as it's data source. 

I'm also working on a mobile app at the same time, but that's more for syncing your roster to your phone so you know what you have when you're not at home.

pabender, 2020-07-16:
@mmillican just throwing out an idea

Could you use the restful api and think of this as more of a microservices architecture

i.e. your app is a microservice that talks to jmri as another microservice?

I think that gets around the need for CORS

mmillican, 2020-07-16:
@pabender Definitely a consideration I had and still a possibility, just was hoping to eliminate that extra hop.

chalabit, 2020-07-17:
I'm also interested in CORS.  I've built an apache website that allow me to track and manage various items on the layout (outsite of JMRI).  I"m attempting to imbed the web-throttle as an iframe inside one of my pages.  This page will also display information from my web app along-side the throttle.  This requires CORS to be enabled.    I may be way-off track, but does JMRI use an angular server?   https://www.npmjs.com/package/angular-http-server     If so, CORS can be enabled via  "angular-http-server --cors" statement, but I don't know where to place the statement.  Any help would be appreciated.

closed 2021-04-14 by stale[bot]