JMRI/JMRI #8846: Add support for filters to web server

rhwood, 2020-07-16:
The JMRI Web Server needs to be able to support [Filters](https://docs.oracle.com/javaee/6/api/javax/servlet/Filter.html) in a manner similar to its support for servlets so that filters can be added that apply to all requests handled by the web server.

rhwood, 2020-07-16:
This is a task needed to allow #8843 to be addressed.

danielb987, 2020-07-16:
If this is implemented, does it mean that JMRI requires Java EE? Does JMRI run on Java SE today, without the need of Java EE?

rhwood, 2020-07-16:
What does a review of the current code and dependencies suggest to you?

danielb987, 2020-07-16:
> What does a review of the current code and dependencies suggest to you?

I almost don't know what Java EE is so I don't know. I ask because I don't know the answer.

rhwood, 2020-07-16:
JMRI has depended upon and shipped parts of Java EE since at least August 2008 if not earlier.

Whether or not that means JMRI “depends on” Java EE is a purely academic question with zero bearing on requirements levied by users.

Randall

closed 2021-04-14 by stale[bot]