JMRI/JMRI #8848: Default state of sensors not working in LocoNet simulator

PUlvestad, 2020-07-18:
JMRI.4.21.1ish+jenkins+20200717T2220Z+Rdebcab155b

When opening panel in LocoNet simulator, "Unknown" sensors are not changing to "Inactive" as set in the Preferences. Nothing in the system console.

2020-07-18 08:38:44,342 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-07-18 08:38:44,372 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter\JMRI\log\messages.log [main]
2020-07-18 08:38:44,377 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter\JMRI\log\session.log [main]
2020-07-18 08:38:44,387 apps.Apps                             INFO  - PanelPro version 4.21.1ish+jenkins+20200717T2220Z+Rdebcab155b starts under Java 11.0.6 on Windows 10 amd64 v10.0 at Sat Jul 18 08:38:44 MDT 2020 [main]
2020-07-18 08:38:44,558 apps.Apps                             INFO  - Starting with profile EMRA_Sim.3f4378bc [main]
2020-07-18 08:38:44,793 node.NodeIdentity                     INFO  - Using 36a770c4-720f-49f6-bd9c-d7f7cfba9239 as the JMRI storage identity for profile id 3f4378bc [AWT-EventQueue-0]
2020-07-18 08:38:45,227 hexfile.HexFileFrame                  INFO  - SensorManager referenced by port is not an LnSensorManager. Have not set the default sensor state. [main]
2020-07-18 08:38:45,227 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-07-18 08:38:45,232 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-07-18 08:38:45,682 startup.PerformFileModel              INFO  - Loading file C:\Users\Peter\Dropbox\JMRI\panels\MonP Signal Panel (20200112).xml [main]
2020-07-18 08:38:46,822 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT794 [AWT-EventQueue-0]
2020-07-18 08:38:46,822 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT794 [AWT-EventQueue-0]
2020-07-18 08:38:46,822 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT796 [AWT-EventQueue-0]
2020-07-18 08:38:46,822 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT796 [AWT-EventQueue-0]
2020-07-18 08:38:46,837 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT1305 [AWT-EventQueue-0]
2020-07-18 08:38:46,842 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT1305 [AWT-EventQueue-0]
2020-07-18 08:38:46,842 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT1309 [AWT-EventQueue-0]
2020-07-18 08:38:46,842 implementation.AbstractTurnout        WARN  - expected Sensor 1 not defined - LT1309 [AWT-EventQueue-0]
2020-07-18 08:38:51,146 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-07-18 08:38:51,146 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter\Dropbox\JMRI\ [main]
2020-07-18 08:38:51,146 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter\JMRI\EMRA_Sim.jmri\ [main]
2020-07-18 08:38:51,146 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter\JMRI\ [main]
2020-07-18 08:38:51,146 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter\ [main]
2020-07-18 08:38:51,156 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Peter\Dropbox\JMRI\Scripts\ [main]
2020-07-18 08:38:51,850 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2020-07-18 08:39:02,992 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]

JMRI.4.21.1ish+jenkins+20200715T1854Z+R0dc7138ccc on my others computer exhibits the same behaviour. 

JMRI.4.20+Rc7ba8249b performs correctly.

Peter.

devel-bobm, 2020-08-03:
The "simulator" connection (implemented in 'loconet.hexfile' ) has a connection-specific loconet Sensor manager, 'loconet.hexfile.lnSensorManager'.  That connection-specific sensor manager had been instantiated as part of `loconet.hexfile.HexFileSystemConnectionMemo`.  

But commit 715c592385491f5dd3559749c7183d8c5dc02ba5 by @pabender has commented-out the code in `loconet.hexfile.HexFileSystemConnectionMemo` which instantiated that conenction-specific Sensor manager. So the simulated loconet connection implementation falls-back to the Sensor manager used by other loconet connections, found in 'loconet.lnSensorManager'.  And, because it is ASSUMED that such connections have real physical hardware, and that the LocoNet "interrogate" function will work on such connections, that manager does _not_ implement the "default" functionality.

@pabender, what was the impetus for removing the connection-specific Sensor manager for the simulated loconet connection?

devel-bobm, 2020-08-03:
@pabender, "never mind".  My mistaken interpretation of the code!

After further review, the above-mentioned changes are _not_ the source of the problem.

Instead, line 391 of 'jmrit.beantable.SensorTableAction` _only_ updates the "default" sensof value for _Internal_ connections.  It cannot update the default value for any other connection type, including the "Simulated" connection types.

devel-bobm, 2020-08-03:
The GUI presentation of "Defaults"->"Initial Sensor State" is as part of a menu item in the menu bar.  To me, this implies that the setting applies to any/all connections available in the Sensor table, not only the Internal Sensor implementation.

But by my reading, the `jmrit.beantable.SensorTableAction` implementation of this "default value" operation can only affect "Internal" sensors.

I infer from @PUlvestad's issue submission that he would like the Sensor "default" value to apply to Sensors created for "LocoNet Simulator" connections.  This seems like it would be a useful addition to JMRI, and my own usage of the simulated connection could also benefit.  And I have encountered a similar situation with "C/MRI Simulator" connection Sensors, as a default other than "Unknown" would have been handy for some signaling development I worked on.

Perhaps we ought to extend the concept to "all" _simulated_ connections, and all beans which _simulate_ a physical object, including Turnouts and Lights, and perhaps other bean types?

KenC57, 2020-08-04:
Since we have lots of scripts that will do the init as needed, that's what
many have used for decades when simulating layouts. It could be said that
adding that as a simulated option instead would be simpler for many users.
But it should be clear that is only for simulation. Knowing that sensors are
not working (getting updated from the initial unknown) is a critical debug
feature.

devel-bobm, 2020-08-04:
Perhaps it may be more sensible to remove the "Initial State" setting from the pull-down menu and instead make it a "bottom of the window" option, for those conenctions where an "initial state" is appropriate, i.e. Simulated or Internal connections only.  That way it would not show up in tabs for inappropriate connection types.  But what to do for the "All" connections tab?

PUlvestad, 2020-08-04:
Just to note again, this option in the preferences has been working up until 4.21.1. I'm at a work computer so can't double check be I'm sure this option is for simulator only. 

Peter

KenC57, 2020-08-04:
Up until just lately, the option to pick a default was only for Internal
sensors and nothing else. It seems somebody is trying to extend that to all
simulators and finding out about the differences between the simulators for
many connections.

PUlvestad, 2020-10-28:
Still not working since 4.20.
[image]

PUlvestad, 2021-02-01:
Fixed with #9363

closed 2021-02-01 by PUlvestad