JMRI/JMRI #8851: Digitrax Programming Track, JMRI, and LocoBuffer-USB Not Working

dulongjp, 2020-07-19:
I recently installed a new Digitrax Evolution DCS210 command station on my layout and it works fine except I cannot get it to work properly with my laptop, LocoBuffer-USB, and JMRI DecoderPro to use the programming track. I have Programming Mode set to Direct Byte and the Programming Track radio button is selected.

In DecoderPro in the Preferences under Connections, I have the System manufacturer set to Digitrax, the System Connection set to "LocoNet LocoBuffer-USB," Settings set to COM3 for the Serial Port, DCS210 (Evoution Command Station) for Command Station Type, Connection Prefix is L, and Connection Name is LocoNet.

When I start up DecoderPro it starts as usual and I can click the Identify button and see that it slightly moves the locomotive and correctly identifies it. If I click the New Loco button, then I get a programming window and can read, enter, write, and save values. I can also open the Turnout Control panel and throw and close turnouts.

What I cannot do is highlight one of my already defined locomotives, click the Program button, and then edit the profile. An empty small window opens in the upper left of the screen, I can expand the window, but there is nothing there except some very defualt tab options (File, Reset, Window, and Help). The main part of the screen is empty. I am then unable to close this window. If I click on the profile for the new locomotive I just defined, then I get the same blank window.

After I hit the Program button, I not only cannot close the blank window that open, but I cannot close DecorderPro either. I have to use the Task Manager to shut down Java and DecoderPro to close the application.

I installed the newest version of JAVA, ver. 8, build 1.8.0, I downloaded the newest LocoBuffer-USB drivers and installed them, and I downloaded JMRI 4.20 and installed it. Same results.

I verfied the wiring and have continuity from the DCS210 to the programming track.

I then tried the SPROG on my laptop, and get the same results as the LocoBuffer-USB. The SPROG is working fine with DecoderPro on my desktop computer. So I think there is something wrong with my laptop, it is using Windows 10.

I followed the JMRI: Debugging Your Installation instructions and everything checks out OK except Step 6: Can decoders be programmed?

Any clues or suggestions will be helpful.

dulongjp, 2020-07-20:
The problem was with the roster. Even when I did a complete new install, Decoder Pro was still finding the old and apparently incompatible roster folder. I deleted the roster folder. Uninstalled DecoderPro, reinstalled it, and then tried to create a roster entry for a single locomotive. It worked, I was able to close the roster entry, reopen it, modifiy it, and close and reopen it again. 

I then went to me desktop computer and copied the roster folder to a network drive, went back to the laptop and removed the roster folder there, and then copied the desktop's roster folder over to the laptop. That solved the problem. I have my complete roster and I am able to modify locomotive entries.

Thanks to Randy on MR Electronics and DCC forum for the clue!

danielb987, 2020-07-20:
Closing the Issue since the problem is solved.

closed 2020-07-20 by danielb987