JMRI/JMRI #8853: CPE unwanted Signal Mast

sidlo64, 2020-07-20:
[profile.zip](https://github.com/JMRI/JMRI/files/4949554/profile.zip)
[session.log](https://github.com/JMRI/JMRI/files/4949556/session.log)

Issue Type: __Bug__

The Control Panel Editor creates an unwanted Signal Mast

How to reproduce

Start PanelPro
Open Table - Signal Mast -- *table is empty*
Create new Control Panel Editor
Open Item palette
Click tab *Signal Mast*
New unwanted Signal Mast **IF$vsm:AAR-1946:SL-2-high-abs ($ 0003)** is created

PanelPro 4.21.1ish+tata+20200720T1848Z+R0c161e32d7
Java HotSpot(TM) 64-Bit Server VM Oracle Corporation 25.181-b13 mixed mode
Linux 4.15.0-111-generic amd64

sidlo64, 2020-07-20:
Log from production release 4.20

[session.log](https://github.com/JMRI/JMRI/files/4949839/session.log)

2020-07-20 22:05:05,951 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2020-07-20 22:05:53,458 audio.JoalAudioFactory                INFO  - Initialised JOAL using OpenAL: vendor - OpenAL Community version - 1.1 ALSOFT 1.15.1 [Listed Table Generation]
**2020-07-20 22:06:18,872 implementation.SignalHeadSignalMast   WARN  - SignalMast system name should start with IF$shsm but is IF$vsm:AAR-1946:SL-2-high-abs($0003) [AWT-EventQueue-0]**
**2020-07-20 22:06:19,983 implementation.SignalHeadSignalMast   WARN  - Attempting to create Mast from non-existant signal head $0003 [AWT-EventQueue-0]**
2020-07-20 22:06:59,484 audio.DefaultAudioManager             INFO  - Shutting down active AudioFactory [On Start/Stop]
2020-07-20 22:06:59,592 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

petecressman, 2020-08-02:
This issue is fixed in the pending PR #8862.  No extra Signa lMast is created.

closed 2020-12-07 by silverailscolo