JMRI/JMRI #8859: Linked Turnouts in Layouts

ReayGrant, 2020-07-22:
Hi there, I have just updated from 4.19 to 4.20. Since the update I can no longer invert one of the turnouts of a linked pair, in a Layout Window. So one Turnout Circle is set Red when the other is Green. Although the animation of the Turnout legs works as you would expect.
This also used to happen in 4.19 when I was using a remote ipad.
I can't find an obvious way of editing the Layout.xml file to change it. I tried swapping LH for RH, swapping the x,y coordinates for the legs and  updating the matching track segments - and it didn't change the behaviour at all. That really had me scratching my head.
Or would it be easier for me to just unlink the Turnouts in the Turnout Table, and use a script to link them instead?  
I have also noticed that if a turnout sticks then you can't fire it again from the Layout Window when using an ipad. I can't get them to fire again from a Panel Window at all.
Thanks for everything 
Reay

dsand47, 2020-07-23:
By **linked** do you mean **Also Throw Turnout**?  If so, both turnouts have to have invert selected or not selected..  Otherwise a stack overflow occurs when they each continuously try to set the other turnout.

The normal fix is to do the inverting at the turnout table and/or use the continuing route edit option.  It appears that the turnout circle color option does not honor the continuing route feature.

geowar1, 2020-07-23:
> It appears that the turnout circle color option does not honor the continuing route feature.
Actually it does; Check the LayoutTurnout::drawTurnoutControls function. It very clearly uses the continuingSense variable to determine what color to use to draw the control circles.

dsand47, 2020-07-23:
> > It appears that the turnout circle color option does not honor the continuing route feature.
> 
> Actually it does; Check the LayoutTurnout::drawTurnoutControls function. It very clearly uses the continuingSense variable to determine what color to use to draw the control circles.

My test case might be wrong.

[linkedturnouts.xml.txt](https://github.com/JMRI/JMRI/files/4967707/linkedturnouts.xml.txt)

Cliff-Anderson, 2020-10-17:
When using 4.18 to modify an ED Panel turnout marked "Also Throw Turnout" and also checking the "Inverted" box, then the other identified Turnout is automatically also marked as Inverted.  Several attempts to only identify one as inverted have consistently been correctly edited within ED.  
   
This automatic matching of "Inverted" condition for "Also Throw Turnout" pairs does not happen when using 4.20 through 4.21.2 versions.  
   
If a user of 4.20 marks one Turnout as Inverted, then the other Turnout must also me **manually** marked as Inverted before attempting to test the results.  If only one is marked, as shown in the "4_20 Also Throw Turnout - Inverted.xml" Panel file, a runaway loop will eventually cause a stack overflow.  
  
On a Win10 OS, the stack overflow will lead to an incomplete crash leaving some hidden fragment of JMRI still running or at least some indicator that any subsequent attempt to load PanelPro will claim that the program is still running.  Task Manager cannot find the indicator, thus a reboot seems to be the only safe cleanup that I can find.  
   
Both of these attached files us None as the primary connection, so they should be safe in any Config Profile folder.  But unless used with proper breakpoints in your IDE, the 4.20 panel can cause problems.  
   
[4_18 Also Throw Turnout - Inverted.xml.txt](https://github.com/JMRI/JMRI/files/5396489/4_18.Also.Throw.Turnout.-.Inverted.xml.txt)
[4_20 Also Throw Turnout - Inverted.xml.txt](https://github.com/JMRI/JMRI/files/5396493/4_20.Also.Throw.Turnout.-.Inverted.xml.txt)

closed 2021-04-14 by stale[bot]