JMRI/JMRI #8861: Inconsistent sort order roster vs print roster

philguk, 2020-07-23:
When I view a roster in DecodePro I can sort by any of the columns. Sorting by ID is a NUMERIC sort
BUT when I choose to print a roster, the list is sorted by ID, but an ALPHA sort
This means that ID 100 is sorted BEFORE ID 2 when printed!
It would be nice to have more control of the printed output, but at the least a NUMERIC sort would be better than noting!

Thanks

Phil G

philguk, 2020-08-08:
Is this the right way to raise an issue?

Pugwash1, 2020-08-08:
Yes. May I ask which version of JMRI, and panelpro or decoderpro (though I doubt that has anything to do with it)? If I get a chance I will look at it Sun. EST else Mon.

philguk, 2020-08-08:
Pretty certain “the latest prod version” but I’ll double check next time I’m in the loft

And thanks for confirming I did the report in the right place

philguk, 2020-08-08:
OK, its DecoderPro  4.18+R37ad3d0

closed 2021-04-14 by stale[bot]