JMRI/JMRI #8864: WebServer panel not shows OBlock status

sidlo64, 2020-07-25:
OBlock from Control Panel Editor has several state. 
WebServer panel does not show them.

Test CPE editor.

Sensors on CPE editor panel control status Clear, Occupied, Not In Use and Power Error.
Warrant table control status Allocated.

[WebCPE.jmri.zip](https://github.com/JMRI/JMRI/files/4975022/WebCPE.jmri.zip)
[image]

silverailscolo, 2020-10-16:
Thanks for adding the example file, makes it easier to see (not) happening what you describe.
Looking at the code passed to Web Server (`web/js/panel.js`) one would expect that most of the information is available, and when I add a "normal" indicator-track connected to the `os5` sensor instead of the `block-5` OBlock, they show state as you expect (web server at theback):
[image]
I will try to fix by adding the missing info to the widget.

silverailscolo, 2020-10-16:
Got OBlocks to show occupancy, but will look for a smarter way to transfer the sensor names (had to add it to the XML):
```
<indicatortrackicon x="60" y="100" level="7" forcecontroloff="false" hidden="no" positionable="false" showtooltip="true" editable="true" class="jmri.jmrit.display.configurexml.IndicatorTrackIconXml">
      <occupancyblock>block-1</occupancyblock>
      <occupancysensor>os1</occupancysensor>
      <errorsensor>es1</errorsensor>
      ...
```
[image]

silverailscolo, 2020-10-20:
Got ``oblocks`` operating via JSON etc! Now working on some server tests.
[image]

Added an empty and sensor controlled indicator tracks at the bottom. Extra states, different assumptions to handle. Needs a bit more time.

mstevetodd, 2020-11-25:
@sidlo64 Can this issue be closed now that PR #9093 has been merged? That change is in the Jenkins builds.

closed 2020-11-26 by sidlo64