JMRI/JMRI #8867: The "MOVE" Option

JDWR, 2020-07-26:
I have noticed a difference in what happens when using the 'Move' option in the Turnout Table as compared to the Block Table. Within the turnout Table the option moves the User Name into the new record previously created but in the Block Table this does not happen and the User Name has to be manually entered into the new record.

Is this a glitch? Is there a similar problem when using this option with the other Tables?

Dave

dsand47, 2020-07-27:
> I have noticed a difference in what happens when using the 'Move' option in the Turnout Table as compared to the Block Table. Within the turnout Table the option moves the User Name into the new [...]

If you look at the system console you will see a Java error.  The Java error is easy to fix.  The side effects are not.

The first problem is that the new block will not have the occupancy sensor, the length and other block attributes.

The track segments and turnouts will lose their block assignments.  This will cause errors for signal masts that were attached to the block boundaries that no longer exist.

Since blocks are strictly internal, there is no value in moving the user name from one internal system name to another internal system name.

The block table has unusual dependencies.  I expect other tables to behave as expected.

Dave Sand

JDWR, 2020-07-27:
Dave,

Once again, thank you for your explanation.

I take it then, that moving all of the record data, its attributes, etc instead of just the User Name will not overcome this problem? All that is changing is the System Name and these have already been embedded in the coding.

Dave

silverailscolo, 2021-02-13:
From the explanation above I conclude that Move (since renamed to "Move User Name" because that is what it does, also when selected from any other column) should not be presented in the Block table.   OK if I code that?

dsand47, 2021-02-13:
> From the explanation above I conclude that Move (since renamed to "Move User Name" because that is what it does, also when selected from any other column) should not be presented in the Block table. OK if I code that?

Removing **Move** from the Block table would be helpful.

Additional background:
- From a panel perspective, only user names are relevant.  This applies to the Layout Editor block assignments and the LayoutBlock objects which connect the panel assignments to the actual blocks.
- From the blocks/paths perspective, only system names are relevant.

icklesteve, 2021-04-05:
Would I be right in thinking that as LightControls do not use NamedBeanHandle ,
the move option should also be dropped from the Light Table?

dsand47, 2021-04-05:
I have not tested this, but I suspect for light controls the move can happen but will not take effect until JMRI is restarted with the updated xml data file.

The previous comments about block attributes probably applies to all of the other beans.

I tested moving a user name from a turnout with feedback sensors to a turnout without feedback sensors.  The feedback attributes were not moved or copied.

KenC57, 2021-04-05:
Dave S mentioned about how the MOVE only moved the username and didn't touch the sensors on a turnout for
feedback. I'd say that makes sense. The MOVE is only dealing with the user name, a label. The feedback in that
case is really a hardware thing of the physical turnout. Now if you did the MOVE of the turnout username and a
matching MOVE of feedback sensor names, as a separate operation, I'd say that's what you wanted.

closed 2021-06-04 by stale[bot]