JMRI/JMRI #8877: change access control on parseOperationsMessage() back from private back to public

Kb0oys, 2020-07-28:
parseOperationsMessage() is a method in the package SimpleOperationsServer, whose purpose is to exchange Operations data over a network.  parseOperationsMessage() parses messages encoded by its shadow method constructOperartionsMessage(); thus, one constructs messages and the other deconstructs those same messages.  Both should have public access control so that clients - such as applications like CATS - can defer to Operations for the structuring of Operations messages.  This decouples the structure of messages from clients.  It did have a public access control until changed by issue 8241.  Please change it back.

danielb987, 2020-07-28:
@pabender 
Could you take a look at this issue? You deprecated this method in commit https://github.com/JMRI/JMRI/commit/a904dc28162fbf14a65e4bb45f946921b35d6c32. @rhwood made the method private in #8241.

pabender, 2020-07-28:
After looking this over, I think the deprecation was correct, but the method shouldn't have been made private yet.

parseOprationsMessage() actually is just decoding a piece of the message after it has been sent over the network.  It doesn't appear to decode the whole message.  parseStatus() really is the method in the server message that actually does something use. 

The missing piece is that there isn't any code in the JMRIClient that handles the operations data.  Until that is in place, parseOperationsMessage() shouldn't have been made private because there isn't any other way to use the message data sent by the server.

closed 2020-08-05 by bobjacobsen

danielb987, 2020-08-05:
@Kb0oys 
The PR #8878 is now merged so the method `parseOperationsMessage()` is now public again.

Kb0oys, 2020-08-05:
Thanks.

Rodney Black

PR #8878 (2020-08-05): Make parseOperationsMessage() public
Closes #8877