JMRI/JMRI #8885: improve unusual requirements for imagelink element in new signal system appearance files

jbnx42, 2020-08-02:
In the process of defining a new signal system and including my custom gif files for some appearances, I ran into what I think are unusual requirements for the image link to custom images. The requirements are spelled out in [Defining Your Own Signaling System](https://www.jmri.org/help/en/html/tools/signaling/DefineNewAspects.shtml) in the section "Create appearance-*.xml files," and the paragraph is the one explaining the imagelink tag contents.

IMHO the requirement for using the preference: prefix or a web link to the JMRI site and not allowing a relative path is less than ideal for the following reasons: (note that I am a Windows user)

- The "preference:" syntax causes the file to not display correctly when trying to view the aspects XML file or the appearance XML file using a local browser such as IE.
- The URL syntax referring to the JMRI site would not work for a layout with no Internet connectivity.
- If someone does not notice this documented requirement and uses an existing appearance XML file as the basis for their own with a relative imagelink URL, they will get unexpected and unpleasant results. The one I found was when adding a Mast (created without incident) at a Turnout. Clicking Done in the "Set Signal Masts at Turnout" dialog caused the Dialog Box to hang due to an error:

2020-08-02 10:55:51,454 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-08-02 10:55:51,495 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\John\JMRI\log\messages.log [main]
2020-08-02 10:55:51,497 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\John\JMRI\log\session.log [main]
2020-08-02 10:55:51,526 apps.Apps                             INFO  - PanelPro version 4.20+Rc7ba8249b starts under Java 14.0.1 on Windows 10 amd64 v10.0 at Sun Aug 02 10:55:51 EDT 2020 [main]
2020-08-02 10:55:54,178 apps.Apps                             INFO  - Starting with profile My_JMRI_Railroad.3e5444f2 [main]
2020-08-02 10:55:54,422 node.NodeIdentity                     INFO  - Using jmri-mCKo0bCtWhNiaaIkLcNok2-3e5444f2 as the JMRI storage identity for profile id 3e5444f2 [AWT-EventQueue-0]
2020-08-02 10:55:56,022 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-08-02 10:55:56,067 hexfile.LnHexFilePort                 INFO  - LocoNet Simulator Started [LocoNet HexFileFrame]
2020-08-02 10:55:57,318 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-08-02 10:55:57,319 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\John\JMRI\My_JMRI_Railroad\ [main]
2020-08-02 10:55:57,320 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\John\JMRI\My_JMRI_Railroad\ [main]
2020-08-02 10:55:57,321 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\John\JMRI\ [main]
2020-08-02 10:55:57,322 util.FileUtilSupport                  INFO  - File path home: is C:\Users\John\ [main]
2020-08-02 10:55:57,323 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-08-02 10:55:58,848 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2020-08-02 10:57:28,153 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.IllegalArgumentException: input == null!
	at java.desktop/javax.imageio.ImageIO.createImageInputStream(ImageIO.java:339)
	at jmri.jmrit.catalog.NamedIcon.<init>(NamedIcon.java:98)
	at jmri.jmrit.display.SignalMastIcon.loadIcons(SignalMastIcon.java:128)
	at jmri.jmrit.display.SignalMastIcon.getIcons(SignalMastIcon.java:100)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:73)
	at jmri.jmrit.display.SignalMastIcon.setSignalMast(SignalMastIcon.java:90)
	at jmri.jmrit.display.layoutEditor.LayoutEditorTools.setSignalMastsDonePressed(LayoutEditorTools.java:9295)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
	at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6636)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6401)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5012)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4844)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4919)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4548)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4489)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2764)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4844)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

csc-sample-student, 2020-08-03:
A partial reply:
 - PR #8886 should improve that huge traceback, replacing it with a somewhat-useful message as two what went wrong
 - JMRI has an [internal XML catalog](https://www.jmri.org/help/en/html/doc/Technical/WebSite.shtml) so that it's own references work fine.  No internet connection is needed. (We think about 10% of JMRI instances don't have an internet connection, including some pretty impressive layouts)

But I'm not sure what to do instead of using the `preference:` prefix.  We use that because JMRI doesn't have any particular static location where any particular set of preferences can be found.  The relative accessing can be used for the shipped signal systems because we know both the signal XML files and the icon files have a fixed relationship within the distributed program.  That relative relationship may not exist for user created files.

Any thoughts on how to do this better?

jbnx42, 2020-08-04:
Thanks for the quick replies and edits!

I think Bob's code in #8886 *DOES* address the 

As to the Internal XML Catalog, I was unaware of that feature. Now that I know about it, it makes sense and explains how JMRI gets its DTD Schemas and hence its parsing abilities when offline! I think the JMRI Help System files that touch on creating your own XML files should mention this feature so that XML file authors can use the http://jmri.org/ URLs in their own data without fear in an offline situation.

jbnx42, 2020-08-04:
Closed in error.

closed 2021-04-14 by stale[bot]