JMRI/JMRI #8888: Missing F29 to F31/F32 in Label & Media tab (ESU Loksound 5 & Lokpilot5 at least) in JMRI v4.20

alain355, 2020-08-03:
Dear team

As I said in groups.io messages https://groups.io/g/jmriusers/message/178031 , when opening the **Label & Media** pane to fill label for function keys F29 to F31/F32, the relevant rows didn't exist. This error applies both to Win 10 or Linux Mint OS.

I tried to find in "C:\Program Files (x86)\JMRI\xml" the file defining this pane wo success.

The error applies also to Lokpilot 5, and maybe to other decoder definitions with more than 29 F keys

dheap, 2020-08-03:
See my response at https://groups.io/g/jmriusers/message/178065

dheap, 2020-08-03:
I've done some preliminary work.

dheap, 2020-08-03:
@alain355 
> I tried to find in "C:\Program Files (x86)\JMRI\xml" the file defining this pane wo success.

You won't find it. the Function Labels pane is currently a special pane defined entirely in JMRI code.

dheap, 2020-08-03:
@dheap 
> The Function Labels pane is currently a special pane defined entirely in JMRI code. I'd like to see that changed, for other reasons.

ESU decoder store some Function Pane settings in CVs and are readable. (But no Function Label text.)

alain355, 2020-08-03:
Thanks Dave, I've seen and answered your message in groups.io. It's not urgent at all.

Take care.

dheap, 2020-08-06:
I'm working on this...

alain355, 2020-08-07:
Thanks Dave.

dheap, 2021-03-16:
Resolved by #8932

closed 2021-03-16 by dheap

alain355, 2021-03-17:
Thanks Dave