JMRI/JMRI #8898: Additional feature suggestion for DecoderPro search bar

bobjacobsen, 2020-08-06:
(See #8883 for background)

Some suggestions from @danielb987 for improvements to the DecoderPro search bar:

- [ ] L&Fs don't always clearly highlight the focused input component. One idea is to highlight the text that explains what the item does. If I search for "Speed" and the checkbox "Remember current speed" is hit, you could give the text "Remember current speed" some different background color. And as soon as the user moves focus or edit the control, the background color goes away. But it would give the user a clear note about which control is found.

- [ ] When the search string can't be found it currently beeps.  Suggest a message box with a message that the string "Abc" cannot be found.

- [ ] If I'm on the search bar and press `<Ctrl>-F`, the search bar goes away, which is great. But if I'm not at the search bar, I would like the cursor to move to the search bar. If I do a search, and hits the the wrong control, I would like to continue the search so I press `<Ctrl>-F` again. But then the search bar disappears and then I press `<Ctrl>-F` again, the search bar is empty. (I'm a keyboard guy, I use the keyboard more that the mouse).

- [ ] Another thing that I'm thinking of is that if the pane is larger than the visible area and you have scroll bars, the found item may not be visible. It would be great if the scrollbars could be moved so that the found item is visible. For example, ESU Loksound 5, pane "Sound levels", has lots and lots of controls on the pane and at the bottom of the pane is "Gear shift sound".

- [ ] When I do a search and it hits a slider with the label on the left of the slider, the text box gets focus. But if it hits a slider with the label on the right of the slider or on the top of the slider, the slider gets focus. It's confusing, but not a big problem. If it's difficult to fix or requires lots of work, you can leave it as is.

- [ ] It would be really nice with search options. @dheap mention searching Pane names. I did a search where the hit was one option of a combo box (Loksound 5, pane "Function Outputs", item "AUX15 Mode", option "Trigger smoke chuff") when I did a search for "chuff". So one option would be if I want to search:
* Pane names
* Labels
* Combo box items
* Internal checkboxes (I don't know what to call this, but if you select Loksound 5 decoder, pane "Function Map", and click the button "Change" on the column "Physical Outputs (wires)", you get a dialog with a lot of checkboxes).

I think it would be good if these options are checkboxes so that you could select all or only some of them.

- [ ] Another option would be if I want to start the search:
* From the first control on first pane (which I suppose is the case now)
* From the first control on the current pane
* From the current control on the current pane (the control that last had the focus)

danielb987, 2020-08-06:
> Another option would be if I want to start the search:
> * From the first control on first pane (which I suppose is the case now)
> * From the first control on the current pane
> * From the current control on the current pane (the control that last had the focus)

If this is implemented, an additional checkbox would be good:
* Search only this pane

If I expect the item to be on the current pane, it would be nice to limit the search to this pane so the focus doesn't move to another pane by misstake.

KenC57, 2020-08-06:
What I've seen in some browser search features are a counter like 1/15 to
say it is focused on the 1st of 15 found matches. If you see 0/0, that meant
the search does not match anything. Alternate to beep or popup for nothing
found.

Other thought is sometimes having case matching would help when you know
exactly what you are looking for. But many other systems seem to have
dropped that option a while ago.

bobjacobsen, 2020-08-06:
To improve I18N, do the text comparison with `org.apache.commons.lang3.StringUtils.containsIgnoreCase()` instead of through `toUpperCase()`

bobjacobsen, 2020-08-06:
The point about highlighting the associated label with focus is on a component might be a good general improvement, not just for search.  Doing it via `WatchingLabel` mechanism might make this straightforward for many of the cases.

dheap, 2020-08-10:
Is there a keyboard shortcut for Search Backwards? `cmd-G` works for Search Again Forward.

bobjacobsen, 2020-08-10:
@dheap Would it help to change -G to repeat the last search? That way if you’re searching backwards, you also hit -G to repeat. That’s an easy add.

dheap, 2020-08-10:
@bobjacobsen That would be one way. I tend to lean towards a keyboard-only solution. That's my personal bias, but it is also probably better for those using assistive technologies.

How about:

-  `(cmd/ctrl)-G` for search forward.
-  `(cmd/ctrl)-shift-G` for search backwards.

A Google search for "search backwards shortcut" came up with that one as the first hit (used by FileMaker Pro).

dheap, 2020-08-10:
@bobjacobsen 
 
> * `(cmd/ctrl)-G` for search forward.
> * `(cmd/ctrl)-shift-G` for search backwards.

Both are also in my favourite BBEdit!

closed 2021-04-14 by stale[bot]