JMRI/JMRI #8899: B&O 1980 Signal System Update

gpmccartney, 2020-08-06:
Updated version of the B&O 1980 signal system with several errors corrected for inclusion in the next release.
[B&O-1980.zip](https://github.com/JMRI/JMRI/files/5036609/B.O-1980.zip)

devel-bobm, 2020-08-06:
@gpmccartney, I see several shortcomings in these changes:

- 'signals/B&O-1980/index.shtml' file does not include the new mast types in the "Signal Mast definitions" list.
- 'signals/B&O-1980/index.shtml' file does not include any description of why short-braking-distance signals would be used.
- Individual short-braking-distance mast `appearance-*-shortblock.xml` files do not include any user-visible description of why short-braking-distance signals would be used.

Please re-submit the zip file once you resolve the problems.

gpmccartney, 2020-08-06:
Okay I have fixed the issues you found.  Here is the corrected zip.  Thanks for checking over it.
[B&O-1980.zip](https://github.com/JMRI/JMRI/files/5038464/B.O-1980.zip)

devel-bobm, 2020-08-07:
I notice that all of the "CPL" mast types define an "unlit" aspect, but "unlit" is not defined in aspects.xml.

I notice that both the permissive mast types define an unlit aspect but use the term "dark" in the image filename, while the other mast types use "unlit".  Since you are using some existing files, we'll leave it as-is.

I notice that the other existing signaling systems which define an "unlit" aspect uses the name "Unlit" (capitalized first letter).  I'd prefer that all "aspect" definitions be consistently capitalized.  

(Sorry for not noticing these earlier.  I'm peeling the onion!)

gpmccartney, 2020-08-07:
No problem.  Will make corrections and upload tomorrow.  Keep peeling!!!

gpmccartney, 2020-08-07:
Okay Bob, Have a go at this one and see if you see any additional corrections.  Thanks.
[B&O-1980.zip](https://github.com/JMRI/JMRI/files/5043870/B.O-1980.zip)

devel-bobm, 2020-08-08:
In most-recent appearance-CPL-High-NJInternational.xml, in the appearance table, you have:
```
    <aspectMapping>
      <advancedAspect>Medium Approach Slow</advancedAspect>
      <ourAspect>Approach Medium</ourAspect>
	  <ourAspect>Approach Slow</ourAspect>
      <ourAspect>Slow Approach Slow</ourAspect>
	  <ourAspect>Restricting</ourAspect>
    </aspectMapping>
```
But this mast does not define "Approach Slow" so the `<ourAspect>` value of "Approach Slow" is not valid.

I would think that "Approach" could protect "Medium Approach Slow", plus any other less-restrictive indication than "Approach"  that can be displayed by this mast.

Please consider and advise.

gpmccartney, 2020-08-08:
You are correct.  Approach slow should not be there.  I'm glad your helping me go through these.  A second set of eyes on this is so helpful.  It all starts running together after a while.   In this case slow approach slow would be the next in line.  I suppose approach would work as well, but would never show unless Slow approach slow was disabled for some reason.  The NJ International mast is the only one I pre defined the markers as that was how it was sold.  Plus it allowed for the markers to be consolidated into a single head.  The other cpls expect 3 heads whether even if there were no upper or lower markers.  A virtual head would need to be created as a place holder.  I think this was the method that was used in the AAR 1946 set as well. 

Greg

devel-bobm, 2020-08-09:
I pulled `<ourAspect>` from a few locations in `appearance-CPL-High-NJInternational.xml`and `appearance-CPL-High-NJInternational-shortblock.xml`.  Left remaining `<ouraspect>` values unchanged.  Should get further into validation, and perhaps thru.

gpmccartney, 2020-08-09:
Okay.  Let me know if you find anything else.  I just updated the xmls I have.

Greg
[B&O-1980.zip](https://github.com/JMRI/JMRI/files/5046758/B.O-1980.zip)

devel-bobm, 2020-08-09:
Automated checks now pass.

gpmccartney, 2020-08-09:
Awesome.

PR #8906 (2020-08-09): B-n-O 1980 signal system submitted by gpmccartney
As submitted by @gpmccartney via issue #8899.

Closes #8899,
closed 2020-08-09 by devel-bobm