JMRI/JMRI #8908: Looking for instances of “../Sidebar.shtml”

jerryg2003, 2020-08-08:
I discovered a couple of help files that “include” the Sidebar from an upper directory.  Usually, this means that when displayed, the links in the sidebar (properly done as relative links) don’t work (get 404s).  I can’t get the search in GitHub (via the browser, which is what I use) to show me all files that have the construct “../Sidebar.shtml” - is there some way to get a list of files that have that so I can investigate them?

Thanks, Jerry

icklesteve, 2020-08-09:
Hi @jerryg2003 
I'm unsure how to do this online, however if you've got a local copy of the code would suggest Notepad++

Open Notepad++
Ctrl + F to open search popup, opens in default to find in single file tab.
Select Find in Files tab, 
Enter search text, eg "../Sidebar.shtml"
Set the directory to a high enough level to capture what you are looking for, in this case "C:\Users\steve\Documents\GitHub\JMRI\help\"
Click Find All, and all instances are displayed in lower part of screen, you can double click on them to open the file up in Notepad++ for html editing.

[image]

Hope this helps,
Steve.

bobjacobsen, 2020-08-09:
These things are definitely easier with a local checkout:
```
% grep -r '\.\./Sidebar\.shtml' help/
help//en/html/tools/uss/Sidebar.shtml:<!--#include virtual="../Sidebar.shtml" -->   
help//en/html/tools/tracker/Sidebar.shtml:<!--#include virtual="../Sidebar.shtml" -->   
help//en/html/tools/consisttool/Sidebar.shtml:        <!--#include virtual="../Sidebar.shtml" -->   
help//en/html/tools/fastclock/Sidebar.shtml:        <!--#include virtual="../Sidebar.shtml" -->   
help//en/html/tools/programmer/Sidebar.shtml:<!--#include virtual="../Sidebar.shtml" -->
help//en/html/tools/signaling/interlock/index.shtml:    <!--#include virtual="../Sidebar.shtml" -->
help//en/html/doc/Technical/Help.shtml:     &lt;!--#include virtual="../Sidebar.shtml" --&gt;
help//en/html/doc/Technical/webupdate/UpdatingDocs.shtml:    <!--#include virtual="../Sidebar.shtml" -->
```

(Note another problem:  Technical/Help.shtml accidentally got converted to entities, and probably no longer works as an include)

To get a local checkout, I think all you need is open the command line, go to some directly where you want it, and then

```
git clone https://github.com/JMRI/JMRI.git
```

Or, to download a .zip copy (harder to update, but doesn't need git installed), go to this link:

[https://github.com/JMRI/JMRI/archive/master.zip](https://github.com/JMRI/JMRI/archive/master.zip)

danielb987, 2020-08-09:
@jerryg2003 
If you are using Windows, I suggest using GitHub Desktop, instead of using git directly. It's much easier for new users to use GitHub Desktop.

An introduction to GitHub Desktop for JMRI is here:
https://www.jmri.org/help/en/html/doc/Technical/githubdesktopinto.shtml

It can be downloaded here:
https://desktop.github.com/

jerryg2003, 2020-08-09:
Thanks all for the suggestions and advice.  I’ll look at all those files.  As to my use of Github, I think I’ve figured out what I need to set up multi-file PRs via the browser.  I downloaded Github to my PC but couldn’t easily figure it out so I dropped it for now.  Might try again at some point.

jerryg2003, 2020-08-10:
I have fixed all these in PR #8907.  It seems that in my zeal to remove duplicate copies of Sidebar.shtml from our repository, I went a bit too far.  I was able to correct the missing links (which no one has reported in the 6 months they have been out there) without having to reintroduce these duplicate files by making one relative link an absolute link as follows:

These are references to tools/Sidebar.shtml so put one absolute link in that file:
help//en/html/tools/uss/Sidebar.shtml:   
help//en/html/tools/tracker/Sidebar.shtml:   
help//en/html/tools/consisttool/Sidebar.shtml:           
help//en/html/tools/fastclock/Sidebar.shtml:           
help//en/html/tools/programmer/Sidebar.shtml:

This one references tools/signaling/Sidebar.shtml so needed to put one absolute link in there:
help//en/html/tools/signaling/interlock/index.shtml:    

This wasn't actually a reference but rather part of the text so have fixed that about when to use "../Sidebar.shtml" or not:
help//en/html/doc/Technical/Help.shtml:     &lt;!--#include virtual="../Sidebar.shtml" --&gt;

Thanks for your help, Jerry

jerryg2003, 2020-08-13:
PR #8907 fixes this.

closed 2020-08-13 by jerryg2003