JMRI/JMRI #8922: Switchboard panel does not play nice with MQTT connections

bobjacobsen, 2020-08-11:
The code to load a range of Turnouts or Sensors into a Switchboard panel contains this code:
```
        if (manuPrefix.startsWith("M")) {
            _insert = "+"; // for CANbus.MERG On event
        }
```

Although that's completely reasonable for a MERG CBUS connection (which has names like MT+1), but it's a problem for any other connection that happens to have letter "M".

Perhaps one approach is to check for whether e.g. ppT1  through ppT24 are valid and existing names; if not, instead of having 1-24 refer to MT1 to MT24, drop back and use the 1st through 24th existing system names.

devel-bobm, 2020-08-11:
I believe that SwitchboardEditor.java _already_ has adequate hint information in the "connection" pull-down.  That connection name hint can be used to discover the SystemConnectionMemo, and that information can be used to determine whether the connection is a CAN connection, and that can be used to determine if inserting the "+" is appropriate.

This is an instance of system-specific functionality in the middle of an otherwise system-agnostic tool - something that ought to be avoided in general.  Perhaps some adding a method to existing bean abstract functionality could allow placing this system-specific code where it ought to be - in the CAN code.

silverailscolo, 2021-02-09:
In-class handling of `M` prefix was removed (so Switchboards not working with non-numeric system naming managers for now). Planning to allow a user editable, validated entry field on the editor for the starting item name identifier (the prefix and type letter are set in the combo's).

closed 2021-04-14 by stale[bot]