JMRI/JMRI #8939: How to tune a BLI Paragon 3 Diesel Decoder with JMRI Decoder Pro

jamosb, 2020-08-16:
**Tuning a BLI Paragon 3 Diesel Decoder with JMRI Decoder Pro**

First off early Paragon 3 decoders have two functions turned on by default that you absolutely need to turn off. (Later Paragon 3 diesels have these functions turned off by default).

First go to the JMRI Decoder Pro CV panel to make these adjustments.

1. Auto emergency brake sequel (CV227). If it is set to 2, turn it off either by setting it to 0 or as I have it set to 1 (1 turns on auto air brake sounds on and off when train is started and stopped, which is nice feature that relieves you from doing it manually). With CV227 set to 2 the train will instant brake with a squeal sound when you turn the throttle down a bit fast. Not realistic and not needed especially if you have a consist. Use emergency brake on your throttle if you need to stop instantly in case of a emergency.

2. Slowdown when turning on dynamic brake grid fan sound (CV244). Set this to 22 to turn off slowdown. you will still get the dynamic brake noise when hitting the function, it just wont just instantly stop the train when the dynamic brake sound is turned on.

Next go to the Basic Speed Settings (I stay away from the speed curves due to jerky movements using BLI decoders).

For this example I am using a Baldwin (RF-16A) BF-16A PRR HO Diesel.

I strongly suggest you get a Track speedometer, if you do not have one else you can eyeball speed based on how fast the train travels per inch, but I use the Accutrack II Model Train Speedometer for accuracy and convenience.

My engine clocked at 127 sMPH out of the box using my Speedometer. (My track voltage is set to 15.5 watts as I use a ESU controller and that is the recommended volts) if you use a lower voltage this speed most likely will be lower but not by much. (shouldn't be below 14.25 watts for DCC operations. I know NMRI DCC standards are 12-14 volts but with modern decoders this is way outdated and these decoders can handle easily up to 18 volts with no issues and are a bit quirky at less than 14 volts).  

The rated speed for this Engine is about 65 MPH, so unless you like your train flying off the tracks the following are adjustments that are easy to make using JMRI.

In the Basic Speed Control panel you are going to set the vMid (CV6) and the VHigh (CV5) voltage. 

For my train this being a diesel i wanted the mid range lower so i set vMid to 50 (this gives you more slow speed control which is helpful for Freight engines, for Passenger engines you should set it to about half the top speed). Then I run at max speed and adjust the vHigh on a straight level track get 65 sMPH at my track voltage using programming on main. I found vHigh at 152 takes me to 65 sMPH (yours will vary according to your track voltage). I do believe these decoders turn off BEMF about a third of the way up the throttle so top speed will fluctuate between 64-66 sMPH this is normal based on the load and the gradients of the track.

Hope these tips help you get your BLI Diesels in better order.

Jamos
[Tuning a BLI Paragon 3 Diesel Decoder with JMRI Decoder Pro.pdf](https://github.com/JMRI/JMRI/files/5081432/Tuning.a.BLI.Paragon.3.Diesel.Decoder.with.JMRI.Decoder.Pro.pdf)

jszinger, 2020-08-17:
> My track voltage is set to 15.5 watts as I use a ESU controller and that is the recommended volts

I think you mean volts instead of watts here.

jamosb, 2020-08-17:
> > My track voltage is set to 15.5 watts as I use a ESU controller and that is the recommended volts
> 
> I think you mean volts instead of watts here.

corrected thank you.

jamosb, 2020-09-02:
I also found that setting BEMF Kd to 10 in Motor Settings stops the jerky slow speed on these decoders.

closed 2021-04-14 by stale[bot]