JMRI/JMRI #8943: Possible issue with NCE and JMRI consisting

bobjacobsen, 2020-08-17:
It was mentioned in this [jmriusers thread](https://groups.io/g/jmriusers/message/178677) that there's "been an issue between JMRI and NCE systems for a long time and relates to a philosophy difference between the two. NCE doesn't want you to use the consist address. It wants you to only use the Lead or Rear address and its own command station's Consist Table takes care of deciding which direction is Forward and routing speed and function commands appropriately."

See [this longer writeup](https://sites.google.com/site/markgurries/home/nce-info/nce-consisting) on NCE consisting.

pabender, 2020-08-17:
Two points however relating to this in general:

1) Any time CV19 consists are involved, the decoder decides whether or not it responds to functions sent to the consist address.

Both NCE and Lenz bypass this requirement when system specific functions are used to create the consist.  They do this by automatically routing commands to the either the locomotive address or the consist address when at least some of the addresses in the consist are accessed.  ( On NCE, the command station will not keep track of more than 6 ( I think ) mid addresses, so it can only have special behavior for the addresses it stores.  On Lenz, you can dial up any address in the consist, and functions will be sent to that address while speed/direction are sent to the consist ).

Getting system specific behavior was the reason that consist support was added in the first place.  In both the Lenz and NCE cases, the system behaves differently when the consist address is used to access the consist.  That is system specific behavior of the throttle/CS interaction.  Something to be aware of, but not necessarily something JMRI should address.  In other words, JMRI is acting just like any other throttle would in this particular case.

2) I believe that EngineDriver, which is being discussed in the JMRI users thread, actually does its own consisting in the application, so the JMRI consist tool may or may not come into play.

dheap, 2020-08-17:
> 1. Any time CV19 consists are involved, the decoder decides whether or not it responds to functions sent to the consist address.

Correct.

> Both NCE and Lenz bypass this requirement when system specific functions are used to create the consist. They do this by automatically routing commands to the either the locomotive address or the consist address when at least some of the addresses in the consist are accessed.

- NCE always sends function commands to the selected loco address, irrespective of whether it's in a consist or not.
- If the selected loco address is a lead or rear loco in the consist table, the function commands are also sent to the associated consist address. 
- Speed/direction commands to loco addresses are only intercepted if the address is a lead or rear loco in the consist table. In this case the speed/direction command is rerouted to the associated consist address, with a direction reversal if it is the rear loco. The direction reversal is so that you don't have to remember which of the two possibilities in a consist you (may have created some time ago) was the lead. Forward is always such that the selected loco is 'pulling' the consist.
- Speed/direction/function commands to all other loco addresses (including mid-consist locos) are passed through unchanged. In the case of mid-consist locos, the decoder automatically ignores them.

> ( On NCE, the command station will not keep track of more than 6 ( I think ) mid addresses, so it can only have special behavior for the addresses it stores. On Lenz, you can dial up any address in [...]

That's another common area of misunderstanding. The 4 mid-consist addresses are _only_ used by the NCE Momentum Button on its Pro Cab throttle. They are totally ignored otherwise, so are totally irrelevant to any discussion of consist speed/direction/function commands. There is no limit to the number of locos you can properly control  in an NCE consist.

> Getting system specific behavior was the reason that consist support was added in the first place. In both the Lenz and NCE cases, the system behaves differently when the consist address is used to [...]

JMRI shouldn't be doing anything special for NCE consists. If you just ignore the consist tools and select a Lead or Rear loco address in a standard JMRI throttle, everything will work _exactly_ the same as if the user did so on a physical NCE throttle.
 
> The throttle button in the consist tool creates a throttle such that the speed and direction goes to the consist address and the function commands go to the first address in the list of addresses. This mimics the Lenz and NCE behavior in a system independent way.

This is where things go wrong. You can't mimic the NCE behaviour and try routing speed/direction/function commands in JMRI. The NCE command station reprocesses them and you have a problem.

The only way you'll get the throttle button in the consist tool to work with NCE is to:

1. Have the tool pop up a "Lead or Rear" question to the user.
2. Create a standard JMRI throttle controlling the Lead or Rear address.
3. Let the NCE command station do the reprocessing and don't try to duplicate it.

> 1. I believe that EngineDriver, which is being discussed in the JMRI users thread, actually does its own consisting in the application, so the JMRI consist tool may or may not come into play.

I'm not familiar with EngineDriver.

pabender, 2020-08-18:
> > 1. 
> This is where things go wrong. You can't mimic the NCE behaviour and try routing speed/direction/function commands in JMRI. The NCE command station reprocesses them and you have a problem.
> 

The behavior is not actually replicating the NCE process, and it was never intended to do so.

The behavior is there to allow JMRI to operate a consist on systems without a system specific consist manager*.

On these systems, the consist tool triggers programming CV19, so to operate the consist and the consist speed and direction, you must use the consist address, but the system doesn’t know anything about the locomotives in the consist, so you have to call up the locomotive address to be able to control all the functions.

For Systems that DO have a system specific consist manager, you can use the normal throttle opening process OR you can Open the throttle with the button and then release it and reacquire. 

*The system independent behavior also applies to some systems with consist managers.  For CVP, the EasyDCC command station knows about command station consists, but not about CV19 consists.  Some other systems ( XpressNet is an example ) have an optional consist manager, i.e. some command stations support it, but others do not.

Now, NCE has its own consisting tool, in part because NCE systems have some quirks that other systems don’t have.   Some of those quirks should be handled in the NCE consist manager.

Now, if changes are required, I believe most of those SHOULD be isolated to the NCE consist manager.

For example, if the NCE consist implementation of reverse just reverses the order of the consist list,  The first locomotive in the list is the one selected by JMRI as the “lead address”, so reversing the list causes the consist tool to use that address when the throttle button is selected.   This is how you choose the lead vs rear address.

The dual addresses in the throttle won’t be used if the same address is returned for the consist address and the first address in the list.  It would take some work to make that happen in the NCE consist manager, but it can be done.

Most command station consists ( which the NCE consist manager doesn’t support ) do this already.  

The assumption with advanced consists is that the consist address will be the address stored in CV19, but strictly speaking, it isn’t required by the consist tool, so it could be hidden by a system specific implementation.

dheap, 2020-08-18:
> The behavior is there to allow JMRI to operate a consist on systems without a system specific consist manager*.

I agree and don't think we can easily integrate the NCE Consist Manager with it.

> Now, NCE has its own consisting tool, in part because NCE systems have some quirks that other systems don’t have. Some of those quirks should be handled in the NCE consist manager.
> 
> Now, if changes are required, I believe most of those SHOULD be isolated to the NCE consist manager.

I agree. Since I'm already somewhat familiar with the NCE consist manager code (and thoroughly across NCE behaviour), I'll take a look at making the required changes.

However, I don't know anything about EngineDriver and the original poster's issues there.

dheap, 2021-03-30:
I have been doing some work on this but it's still a WIP.

dheap, 2021-05-14:
Still a WIP on my part.