JMRI/JMRI #8967: CI no longer catching CRLF line ends

bobjacobsen, 2020-08-24:
At one point, we had an Ant task run in CI that caught files with CRLF line ends.  That doesn't seem to be happening now, see PR #8981, #8955 and Issue #8965 

It would be good to restore that.

danielb987, 2020-08-24:
The file `checkstyle.xml` in the root folder has this section:
```
  <module name="RegexpMultiline">
    <property name="format" value="\r\n"/>
    <property name="message" value="CRLF line endings are prohibited"/>
  </module> 
```
Maybe that has failed. Does other checkstyle checks work?

bobjacobsen, 2020-08-24:
The Ant target was `checklineends`:

```
    <target name="checklineends" description="check text files for proper LF line ends (Linux, Mac only)">
        <property name="test.includes" value="jmri.util.FileLineEndingsTest" />
        <antcall target="test-single" />
    </target>
```

but that runs a JMRI JUnit test that's no longer present.

bobjacobsen, 2020-08-24:
There was no CheckStyle report on #8891 nor has there been one on some more recent PRs.  It seems to no longer be running in CI.

danielb987, 2020-08-24:
> The Ant target was `checklineends`:
> but that runs a JMRI JUnit test that's no longer present.

I have that file in one of my branches:
https://github.com/danielb987/JMRI/blob/LogixNG_new_21/java/test/jmri/util/FileLineEndingsTest.java

This branch was latest updated 12 jab 2020.
Another branch that doesn't have the file was latest updated 26 feb 2020.

So this file seems to have disappeared between 12 jan 2020 and 26 feb 2020.

bobjacobsen, 2020-08-24:
```
% git log -- java/test/jmri/util/FileLineEndingsTest.java
commit fbba78c243d4fb20a5727326fd04bc2248f2c3f6
Author: Paul Bender <paul.bender@acm.org>
Date:   Thu Feb 6 19:01:00 2020 -0500

    remove the Java FileLineEndingsTest (replaced by checkstyle check)

commit 5901018ca12cbfe15590bcf36d1556105585e48e
Author: Paul Bender <paul.bender@acm.org>
Date:   Tue Jan 28 19:35:20 2020 -0500

    disable tests designed to fail; move line endings check to checkstyle
```

https://github.com/JMRI/JMRI/commits/fbba78c243d4fb20a5727326fd04bc2248f2c3f6

PR #8025

pabender, 2020-08-26:
according to the script, checkstyle should be running as part of the static analysis:

https://github.com/JMRI/JMRI/blob/74e240ee98a42eb4c55e76a60062d911b604ac38/.github/workflows/static-analysis.yml#L34-L35

I can't verify that works correctly on Windows, which I think is what is used for the static analysis checks.

You can run it directly by calling:

`mvn checkstyle:check`

on one of my linux machines, that flags a few jython files with the wrong line endings.

pabender, 2020-08-26:
Here's the config from the pom file (for what runs in static analysis):

https://github.com/JMRI/JMRI/blob/74e240ee98a42eb4c55e76a60062d911b604ac38/pom.xml#L147-L175

pabender, 2020-08-26:
incidentally, the files I have flagged locally may actually be generated by my IDE and not actual source files.

danielb987, 2020-08-26:
> I can't verify that works correctly on Windows, which I think is what is used for the static analysis checks. The pom file looks correct for that.

When I run `mvn checkstyle:check` on Windows 8.1, I get a **huge** number of errors. I think `git` changes line endings on Windows.

Can this check be moved to Linux instead of Windows on the CI servers?

danielb987, 2020-08-27:
> Can this check be moved to Linux instead of Windows on the CI servers?

It seems that spotbugs is run on a Ubuntu computer so that don't seem to be the problem.

danielb987, 2021-04-14:
I think this issue should remain open since it's affecting the quality of the code and I don't think it's resolved. So I open it again.

danielb987, 2021-05-29:
This issue should be fixed.

danielb987, 2021-11-26:
@bobjacobsen 
Has this been fixed? I believe that you fixed it, but I'm not sure?

bobjacobsen, 2021-11-26:
This is fixed in Gihub ci now.

closed 2021-11-26 by bobjacobsen