JMRI/JMRI #8968: Create Test Release 4.21.1

bobjacobsen, 2020-08-25:
This is the next release in the 4.22 cycle. It's intended to be created from the HEAD of the master branch.

bobjacobsen, 2020-08-25:
The release-4.21.1 branch has been created. 

Maintainers, please set the 4.22.2 milestone on pulls from now on, as that will be the next test release from the HEAD of the master branch.

Jenkins will be creating files shortly at the [CI server](https://builds.jmri.org/jenkins/job/testreleases/job/4.21.1/)

danielb987, 2020-08-26:
@bobjacobsen 
You seem to have closed the wrong milestone. If I select an open PR and try to set the milestone, I can select 4.21.1 but not  4.21.2, since the later is closed.

I have closed 4.21.1 and reopened  4.21.2.

bobjacobsen, 2020-08-26:
Oops. Fixed. Thanks for catching it.

closed 2021-01-31 by silverailscolo