JMRI/JMRI #8969: SPROG 3 Command Station Fails to Open in 4.21.1

crosland, 2020-08-26:
Originally noted by @peter ulvestad

2020-08-23 12:47:56,444 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-08-23 12:47:56,463 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\Peter\JMRI\log\messages.log [main]
2020-08-23 12:47:56,464 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\Peter\JMRI\log\session.log [main]
2020-08-23 12:47:56,476 apps.Apps                             INFO  - PanelPro version 4.21.1ish+jenkins+20200822T1013Z+R6eff6b18b7 starts under Java 11.0.6 on Windows 10 amd64 v10.0 at Sun Aug 23 12:47:56 MDT 2020 [main]
2020-08-23 12:47:56,633 apps.Apps                             INFO  - Starting with profile Sprog_CS.3eacb5cb [main]
2020-08-23 12:47:56,891 node.NodeIdentity                     INFO  - Using 36a770c4-720f-49f6-bd9c-d7f7cfba9239 as the JMRI storage identity for profile id 3eacb5cb [AWT-EventQueue-0]
2020-08-23 12:47:56,955 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "SPROG Command Station" [main]
2020-08-23 12:47:57,191 serialdriver.SerialDriverAdapter      INFO  - COM4 port opened at 9600 baud, sees  DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2020-08-23 12:47:57,253 jmri.InstanceManager                  ERROR - Should not set default of type jmri.CommandStation to null value [main]
2020-08-23 12:47:57,256 managers.ManagerDefaultSelector       WARN  - SystemConnectionMemo for SPROG Command Station (class jmri.jmrix.sprog.SprogSystemConnectionMemo) provides a null interface jmri.CommandStation instance [main]
2020-08-23 12:47:57,258 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.managers.ManagerDefaultSelector: System connection SPROG Command Station provides a null manager for interface jmri.CommandStation [main]
2020-08-23 12:47:57,657 plementation.JmriConfigurationManager ERROR - Exception initializing jmri.util.startup.StartupActionsManager: jmri.util.prefs.InitializationException: Unable to run startup actions due to earlier failures. [main]
2020-08-23 12:47:59,711 sprog.SprogCommandStation             INFO  - Registering with power manager [SPROG slot thread]
2020-08-23 12:48:06,876 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-08-23 12:48:06,877 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\Peter\Dropbox\JMRI\ [main]
2020-08-23 12:48:06,878 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\Peter\JMRI\Sprog_CS.jmri\ [main]
2020-08-23 12:48:06,878 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\Peter\JMRI\ [main]
2020-08-23 12:48:06,879 util.FileUtilSupport                  INFO  - File path home: is C:\Users\Peter\ [main]
2020-08-23 12:48:06,879 util.FileUtilSupport                  INFO  - File path scripts: is C:\Users\Peter\Dropbox\JMRI\Scripts\ [main]
2020-08-23 12:48:07,628 PanelPro.PanelPro                     INFO  - Main initialization done [main]
2020-08-23 12:48:40,863 managers.DefaultShutDownManager       INFO  - Normal termination complete [AWT-EventQueue-0]

Works fine using 4.20.

crosland, 2020-08-26:
Seems that something has changed around the instance manager or connection memo. There is no CommandStation class stored in the connection memo.

@pabender could this be related to recent memo refactoring?

I am a bit out of my depth here. I can make the exception go away by adding code to SprogSystemConnectionMemo.ConfigureManagers() to store a CommandStation instance 

```
        switch (sprogMode) {
            case OPS:
                ThrottleManager sprogCSThrottleManager = new jmri.jmrix.sprog.SprogCSThrottleManager(this);
                jmri.InstanceManager.setThrottleManager(sprogCSThrottleManager);
                store(sprogCSThrottleManager,ThrottleManager.class);
                CommandStation sprogCommandStation = new jmri.jmrix.sprog.SprogCommandStation(st);
                store(sprogCommandStation, CommandStation.class);
                break;

```
but then no packets get to the track, so there's still something missing.

pabender, 2020-08-26:
I'll take a look at it.  There isn't anything immediately obvious about why it isn't working.

The stack trace indicates a null command station is being registered with the instance manager, so something isn't right for sure.

pabender, 2020-08-26:
I think I see the issue, but I'll have to wait to fix it until I have access to hardware.  Should be able to get that done tonight.

crosland, 2020-08-27:
Fixed by #8973

crosland, 2020-08-27:
Thanks for looking at this @pabender

PR #8973 (2020-08-28): restore missing SPROG command station in OpsMode
fixes #8969
closed 2020-08-27 by crosland