JMRI/JMRI #8970: Add Comment field to Logix

jerryg2003, 2020-08-26:
Most tables have a “comment” field to allow users to store some non-specific information about a particular table entry, whether personal notes, reminders, or operational information.  I haven’t found such a field for Logix but think it would be useful to have.  Some Logix can get quite complex and I find it helpful to keep notes describing what the Logix is meant to do, any special issues,  and often just why I created this particular Logix in the first place!  I would like to store that information in the panel file rather than keeping separate notes as I do now.

Suggest adding a field (call it “Description” or just “Comment”) appearing below the User Name Field on the Create Logix page.  This this should allow several lines of text (is there a maximum?).  The page listing all Logix could have a column with a “click here to see the comment” (or not).  The “Browse” function should display the comment also so that it can be printed or saved elsewhere with the Logix.

I don’t think such a field is necessary with each conditional or action.

Thanks, Jerry

dsand47, 2020-08-26:
The Comment field is standard with all of the tables, including Logix.  Right click the table header and make sure that Comment column is checked.

jerryg2003, 2020-08-27:
Thanks Dave.  Found it.   I checked and this wasn’t talked about anywhere in Help, although “right click” did show up in Help for Warrant and Oblock Tables,and In Digitrax help re Turnout Table (None of which I use).  So something else for me to add in a few more Help places...

Btw, I found that when you right-click on header of Sensor table, you get “status” listed three times.  Different statuses?

Jerry

jerryg2003, 2020-09-03:
See PR #8989

closed 2020-09-05 by jerryg2003

icklesteve, 2020-09-05:
> Btw, I found that when you right-click on header of Sensor table, you get “status” listed three times. Different statuses?

Hi Jerry, see https://github.com/JMRI/JMRI/issues/5491

Steve.