JMRI/JMRI #8974: Spotbugs: CheckForNull does not override ParametersAreNonnullByDefault

danielb987, 2020-08-27:
While trying to understand the problem in https://github.com/JMRI/JMRI/issues/6510#issuecomment-680422540, I found that `@CheckForNull` for a parameter does not overrides `@ParametersAreNonnullByDefault` for the class.

```
package jmri;

import javax.annotation.*;

@ParametersAreNonnullByDefault
class ClassA {
    
    public void testA(@CheckForNull Object param) {
        // Do nothing
    }
    
    public void testB(@Nullable Object param) {
        // Do nothing
    }
}
```

This means that Spotbugs will give an error on `new ClassA().testA(null);` but not on `new ClassA().testB(null);`, since the later is annotated with `@Nullable`.

The [JMRI documentation](https://www.jmri.org/help/en/html/doc/Technical/SpotBugs.shtml) about Spotbugs say that `@CheckForNull` can be used for parameters, but that's not true if the class is annotated with `@ParametersAreNonnullByDefault`.

The JMRI documentation says that `@Nullable` shouldn't be used and that `@CheckForNull` should be used instead. But if the class is annotated with `@ParametersAreNonnullByDefault`, `@Nullable` works but `@CheckForNull` doesn't.

My branch [nonnull_test_4](https://github.com/JMRI/JMRI/compare/master...danielb987:nonnull_4_test) has a working example of the above.

danielb987, 2020-08-27:
@bobjacobsen @pabender 
Any comments on this issue?

The JMRI documentation says that `@Nullable` shouldn't be used since SpotBugs doesn't really check anything when this is used. But I don't really understand what Spotbugs should check when `@Nullable` is used, so I don't see the problem with using it.

danielb987, 2020-08-28:
I have done additional testing of this class:
```
package jmri;

import javax.annotation.*;

@ParametersAreNonnullByDefault
class ClassA {
    
    public void testA(@CheckForNull Object param) {
        testC(param);
    }
    
    public void testB(@Nullable Object param) {
        testC(param);
    }
    
    private void testC(Object param) {
        // Do nothing
    }
    
}
```

The annotation `@CheckForNull` doesn't seem to do anything. Spotbugs complains if I call `testA(null)` and Spotbugs does not complains about the call `testC(param)` from the method `testA`. So Spotbugs does not treat the `param` in the method declaration for testA as a parameter that can be null.

The annotation `@Nullable` makes Spotbugs be happy with the call `testB(null)` but Spotbugs complains about the call `testC(param)` from the method `testB`. So Spotbugs clearly treats the `param` in the method declaration for testB as a parameter that can be null.

So my testing shows that `@Nullable` works fine but `@CheckForNull` does not work at all when the class is annotated with `@ParametersAreNonnullByDefault`.

The comment in the [JMRI documentation](https://www.jmri.org/help/en/html/doc/Technical/SpotBugs.shtml) about spotbugs which tells that `@CheckForNull` should be used instead of `@Nullable` was written for FindBugs before we switched over to Spotbugs, so it may be a difference between Spotbugs and Findbugs. But the comment is not valid today.

bobjacobsen, 2020-08-28:
A bit short of time right now, but see java/src/apps/FindBugsCheck.java

closed 2021-04-14 by stale[bot]