JMRI/JMRI #8981: Question about organization of files in the JMRI Git Repository under /help/en/package/

jerryg2003, 2020-08-29:
As I look to update various Help files, I try to follow the principles laid out under "Organization of the Help Files" (https://www.jmri.org/help/en/html/doc/Technical/Help.shtml).  One, in particular, is somewhat vexing: 

"there are several file trees:
package - organized like the source package tree, this contains code-specific help files for e.g. a particular window or component. For example, a window that's created by the jmri.jmrit.speedometer.SpeedometerFrame class (from the src/jmri/jmrit/speedometer/SpeedometerFrame.java file) should have its window-specific help in a file at package/jmri/jmrit/speedometer/SpeedometerFrame.shtml.
html - general descriptions, tutorials, etc. This in turn in organized into directories that represent general areas..."

I find a lot of  "general descriptions" under "package," for example Warrants.shtml (which is actually in the "logix" help directory under package/jmri/jmrit/logix, which placement I don't understand either).  Q1:  is it worth my trying to do some reorganization as I go along to better correspond to the principles?  Certainly will make future maintenance easier, and if we ever develop "breadcrumbs" for Help, make finding things easier.

Q2: should I split and move parts of Warrants.shtml to a file under help/en/html/tools?  [And fix the index and TOC to point there rather than to package/?]

Q3: Related to that, does the current other contents of the package/jmri/jmrit/logix directory make sense?  It contains files for a bunch of tables , but not for the Logix Table (which are under beantable/).

Thanks, Jerry

dsand47, 2020-08-29:
Jerry,

For Q3, I suspect the "logix" directory name is a historical artifact.  The contents certainly have nothing to do with Logix.  The contents are for the OBlocks/Portals/OPaths/Warrants system.

Dave Sand

jerryg2003, 2020-08-30:
Re Q3: since I’m not a code maven, can someone please tell me what the directory name should be so that these files follow the “source package tree”?  I’ll then do the appropriate moving.

Thanks.

dsand47, 2020-08-30:
> Re Q3: since I’m not a code maven, can someone please tell me what the directory name should be so that these files follow the “source package tree”? I’ll then do the appropriate moving.

The help files currently follow the package tree.  They are in the same relative location as the Java code and tests.

I don't see anyone taking on the challenge of moving the tool from jmri/jmrit/logix to something like jmri/jmrit/warrants.  That is a lot of work to fix a little confusion.

Currently in the TOC "Warrants" and "Warrants Introduction" are the same file: Warrants.shtml.  It might be reasonable to create a condensed version of Warrants.shtml at html/tools for use at the "Warrants" entry in the TOC.

@petecressman Do you have any thoughts on this issue?

petecressman, 2020-08-30:
Sure, it makes sense.I apologize for having used that directory for the development of OBlocks and warrants.  The history is that The first work I did in JMRI was with Logix. After the first phase was done, it was my intention to rewrite Logix with a more OOPs design.  But some tentative experiments with animated trains got serious and before I ever got going on the project.
I have missed any conversations about a directory change and am unaware that it has been requested. Since things have become formal in JMRI, it certainly would be much better to have the warrant code in a warrant package.  Should the OBlock, Portal, OPath code be separate also? where? Perhaps beantable.oblock? What is Q3? Should I know about Q1 & Q2?

I will do the move. Let me know if jmri/jmrit/beantable/oblock is OK or if it should be something else..
Pete C.

dsand47, 2020-09-01:
Pete,

Q1 and Q2 were about the organization of the Warrant help files.  You can see the original questions on GitHub, issue 8981.

The only issue with the "logix" directory name is that it confuses people from time to time.

I don't see any issue with beantable.oblock.  

According to my understanding of the various JMRI packages, ideally the bean and manager classes should not be in jmri.jmrit.*.  Moving OBlocks/OPaths/Portals sound like a lot of work with no current value.

Dave Sand

jerryg2003, 2020-09-01:
Certainly no one should be bothering with “a lot of work with no current value” when there is so much else to do.  I’ll do a little work with some current value as Dave suggests to create a help file under html/tools as the intro to warrants and make appropriate changes to TOC and index.  That way we will have general information in the html tree and the source package info in the package tree (where it already is), as the guidance on organizing help files states.

Jerry

jerryg2003, 2020-09-03:
@petecressman :
I'm well on the way to having a WarrantsOverview.shtml to put in the tools directory.  I am confused by one section of the current Warrants.shtml:

"OBlocks, Portals and OPaths can be used with Layout Editor.  Compatibility is   achieved by using the same occupancy detection sensors for LayoutBlocks as for OBlocks. If there is no corresponding  Control Panel layout diagram, Circuit Builder cannot be used. The Occupancy Block Tables will have to be used to define the  OBlocks, Portals and Paths."

I understand that if you use the same occupancy sensors in Layout Editor Blocks as you use in CPE Oblocks, that's fine and good, but I don't understand the first sentence at all.  I don't see how OBlocks, Portals, and OPaths are used with Layout Editor.

Please advise.
Thanks, Jerry

petecressman, 2020-09-03:
I see how there might be confusion by the sentence.  It may give the impression that "used with" means "used by".  That is definitely not the case.  The sentence should be cleaned up to only say the same occupation detection can be done by both Layout Blocks and OBlocks. i.e. there is no conflict between the two approaches.

I don't know of specific layouts where this is done.  However, it is possible that someone might have a CPE panel with a CTC screen or perhaps a duplicate iconic representation of the LOE panels.  I agree a presentation with no CPE panels is farfetched. But it is conceivable that warrants might be run by scripts in this case.

I am relieved to be able to take a pass on moving the logix package.  It saves me from having to be some rather tedious work.  Also, it avoids having to be sure I do deprecation properly so as not to louse up warrant scripts.
Thanks for your work,Best Regards,Pete C

closed 2020-09-11 by jerryg2003