JMRI/JMRI #8983: BUG in TurnoutStatePersistence.py jaython program

forty-ford, 2020-08-30:
The program works fine except when debug is on!  I'm trying to use it make a block persistence program but when the debug is turned on a core dump is caused by Line #71. (if my memory serves me correctly)  I haven't tried it again in a couple of weeks.
Thanks,
Donn

dsand47, 2020-08-30:
Donn,

Is there something missing in the built-in block persistence?

Dave Sand

forty-ford, 2020-08-30:
I'm not aware of the built-in block persistence.  When I shut down the layout I want to save the train names located in a block.  That way after an operating session I can come back to those settings.  It would be great if that has already been created and I just haven't found it.
Thanks,
Donn

dsand47, 2020-08-30:
Donn,

Look for a blockvalues.xml file.

I am assuming that you have a Layout Editor panel with blocks assigned to the blocks.  if not, then automatic block tracking will not work.

Dave Sand

forty-ford, 2020-08-30:
I do. Thanks for the tip. I will search for the file, and get back :)
Donn

forty-ford, 2020-08-30:
I found several.  The last created by 3.11.2 in Jan. of 2015.  Looks like I forgot about this file, and don't remember how it was created...  I do remember using it before, but it's been a long time.  How did I create it?
Donn

forty-ford, 2020-08-30:
Thanks Dave, I have found the references for creating the file.
Donn

closed 2020-08-30 by forty-ford