JMRI/JMRI #9003: Java exception clicking "Add" for new Sensor

jerryg2003, 2020-09-07:
JMRI 21.1.  Loaded with SPROG Sim and CMRI Sim.  No panel file.  Using Sensor Table window loaded as a start-up action or just opening it from Tools->Tables, when I click "Add" on the empty Sensor Table I get:

2020-09-07 11:36:11,859 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.beantable.AbstractTableAction.configureManagerComboBox(AbstractTableAction.java:244)
	at jmri.jmrit.beantable.SensorTableAction.addPressed(SensorTableAction.java:134)
	at jmri.jmrit.beantable.AbstractTableAction$1.lambda$0(AbstractTableAction.java:79)
	at javax.swing.AbstractButton.fireActionPerformed(Unknown Source)
	at javax.swing.AbstractButton$Handler.actionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.fireActionPerformed(Unknown Source)
	at javax.swing.DefaultButtonModel.setPressed(Unknown Source)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(Unknown Source)
	at java.awt.Component.processMouseEvent(Unknown Source)
	at javax.swing.JComponent.processMouseEvent(Unknown Source)
	at java.awt.Component.processEvent(Unknown Source)
	at java.awt.Container.processEvent(Unknown Source)
	at java.awt.Component.dispatchEventImpl(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.processMouseEvent(Unknown Source)
	at java.awt.LightweightDispatcher.dispatchEvent(Unknown Source)
	at java.awt.Container.dispatchEventImpl(Unknown Source)
	at java.awt.Window.dispatchEventImpl(Unknown Source)
	at java.awt.Component.dispatchEvent(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.awt.EventQueue$4.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)

icklesteve, 2020-09-08:
Unable to reproduce in current HOM with a new CMRI Sim / SPROG Sim profile.xml

```
<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.cmri.serial.sim.configurexml.ConnectionConfigXml" disabled="no" manufacturer="C/MRI" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="19200" systemPrefix="C" turnoutInterval="250" userName="C/MRI">
            <options/>
            <polllist pollseq=""/>
        </connection>
        <connection xmlns="" class="jmri.jmrix.sprog.simulator.configurexml.ConnectionConfigXml" disabled="no" manufacturer="SPROG DCC" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="" systemPrefix="S" turnoutInterval="250" userName="SPROG Sim">
            <options>
                <option>
                    <name>OperatingMode</name>
                    <value>SPROG Programmer</value>
                </option>
            </options>
        </connection>
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd"/>
</auxiliary-configuration>
```

Also tried with MERG CBUS Sim / CMRI Sim / SPROG Sim combinations, all OK when clicking Add from empty Sensor Table.

jerryg2003, 2020-09-08:
Here is the profile.xml file from my profile directory.  I just tried it again and got same exception.  JMRI 4.21.1+R90f7b8bfd, Java 1.8.0_241 (en).  Let me know if you need to see anything else.  

Note several CMRI nodes defined (no change from when I was using 4.20):
<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.sprog.simulator.configurexml.ConnectionConfigXml" disabled="no" manufacturer="SPROG DCC" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="" systemPrefix="S" turnoutInterval="250" userName="SPROG Sim">
            <options>
                <option>
                    <name>OperatingMode</name>
                    <value>SPROG Programmer</value>
                </option>
            </options>
        </connection>
        <connection xmlns="" class="jmri.jmrix.cmri.serial.sim.configurexml.ConnectionConfigXml" disabled="no" manufacturer="C/MRI" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="19200" systemPrefix="C" turnoutInterval="250" userName="C/MRI">
            <options/>
            <polllist pollseq="1,2,3,4,5,6,11"/>
            <node name="1">
                <parameter name="nodetype">2</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2222211000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="2">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="3">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="4">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="5">
                <parameter name="nodetype">2</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2110000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">0000000000000000</parameter>
[... 25 lines cut ...]
    </connections>
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
        <perform xmlns="" class="apps.configurexml.CreateButtonModelXml" name="jmri.configurexml.LoadXmlConfigAction" type="Button">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="jmri.jmrit.beantable.LightTableAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
[... cut ...]
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="jmri.jmrit.jython.JythonWindow" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="apps.SystemConsoleAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
    </startup>
</auxiliary-configuration>

jerryg2003, 2020-09-08:
hmmm....That didn't come thru, did it?

 ```<?xml version="1.0" encoding="UTF-8"?>
<auxiliary-configuration xmlns="http://www.netbeans.org/ns/auxiliary-configuration/1">
    <connections xmlns="http://jmri.org/xml/schema/auxiliary-configuration/connections-2-9-6.xsd">
        <connection xmlns="" class="jmri.jmrix.sprog.simulator.configurexml.ConnectionConfigXml" disabled="no" manufacturer="SPROG DCC" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="" systemPrefix="S" turnoutInterval="250" userName="SPROG Sim">
            <options>
                <option>
                    <name>OperatingMode</name>
                    <value>SPROG Programmer</value>
                </option>
            </options>
        </connection>
        <connection xmlns="" class="jmri.jmrix.cmri.serial.sim.configurexml.ConnectionConfigXml" disabled="no" manufacturer="C/MRI" port="" reconnectMaxAttempts="100" reconnectMaxInterval="120" speed="19200" systemPrefix="C" turnoutInterval="250" userName="C/MRI">
            <options/>
            <polllist pollseq="1,2,3,4,5,6,11"/>
            <node name="1">
                <parameter name="nodetype">2</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2222211000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="2">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="3">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="4">
                <parameter name="nodetype">1</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2210000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">1000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
            </node>
            <node name="5">
                <parameter name="nodetype">2</parameter>
                <parameter name="bitspercard">24</parameter>
                <parameter name="transmissiondelay">0</parameter>
                <parameter name="num2lsearchlights">0</parameter>
                <parameter name="pulsewidth">500</parameter>
                <parameter name="locsearchlightbits">000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cardtypelocation">2110000000000000000000000000000000000000000000000000000000000000</parameter>
                <parameter name="cmrinetoptions">0000000000000000</parameter>
                <parameter name="cmrinodedesc"/>
[... 25 lines cut ...]
    <startup xmlns="http://jmri.org/xml/schema/auxiliary-configuration/startup-4-3-5.xsd">
        <perform xmlns="" class="apps.configurexml.CreateButtonModelXml" name="jmri.configurexml.LoadXmlConfigAction" type="Button">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="jmri.jmrit.beantable.LightTableAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="jmri.jmrit.beantable.MemoryTableAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
[... cut ...]
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="jmri.jmrit.jython.JythonWindow" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
        <perform xmlns="" class="jmri.util.startup.configurexml.PerformActionModelXml" name="apps.SystemConsoleAction" type="Action">
            <property name="systemPrefix" value=""/>
        </perform>
    </startup>
</auxiliary-configuration>
 ```

silverailscolo, 2020-09-08:
Jerry, can you attach a zip of the panel xml file?

jerryg2003, 2020-09-08:
This happens before I load any panel xml file.  I open JMRI, it completes its startup (including opening the Sensor Table and system console windows, among others).  I click “Add...” and the Java exception shows up in the System Console.
________________________________
jerryg2003@aol.com

dsand47, 2020-09-08:
Try renaming the user-interface.xml file.  It is located in the profile/<randomlookingname> directory.

If that resolves the problem, that means there is a bad or unexpected entry in the file.  Upload the file for review.

jerryg2003, 2020-09-09:
Now we are getting somewhere!  JMRI worked OK when I renamed the user-interface.xml from the directory profile/[long string of characters]/.  Attached here for your perusal with additional surname of .txt is the file that causes the Java exception:
[user-interface.xml.txt](https://github.com/JMRI/JMRI/files/5191872/user-interface.xml.txt)

As I look at this file and compare to some earlier copies, I see the following lines:

    <comboBoxLastValue xmlns="http://jmri.org/xml/schema/auxiliary-configuration/combobox-4-3-5.xsd">
        <comboBox xmlns="" lastSelected="MQTT Connection" name="MQTT"/>
        <comboBox xmlns="" lastSelected="MQTT" name="jmri.jmrit.beantable.SensorTableAction.SystemSelected"/>
        <comboBox xmlns="" lastSelected="MQTT" name="jmri.jmrit.beantable.TurnoutTableAction.SystemSelected"/>
        <comboBox xmlns="" lastSelected="localhost" name="jmri.jmrix.mqtt.MqttAdapter.hostname"/>
    </comboBoxLastValue>

I did previously create an MQTT connection and created some MQTT sensors.  On a subsequent opening of JMRI, I deleted the MQTT connection but it looks like it was still remembered here.  Is it possible that JMRI is confused on trying to open the sensor "Add" window in looking at these lines in  user-interface.xml and trying to put MQTT in the comboBox when there is no MQTT connection?  If so, shouldn't it delete these lines if the connection is deleted? [Just my novice musings, of course...]

Jerry

jerryg2003, 2020-09-09:
Further confirmation of the above:

I went back to the user-interface.xml that failed (with the MQTT lines in it).  Opened JMRI, created an MQTT connection, went to add a sensor and (as expected) no exception.  I then added a CMRI sensor (thus resetting the comboBoxLastValue to CMRI), deleted the MQTT connection, saved, and restarted JMRI.   No more exception on trying to create a sensor (btw, same behavior for lights and turnouts).

Hope this helps in tracking it down.
Jerry

PR #9521 (2021-03-14): Prevent NPE in Create new Bean Manager ComboBox
In AbstractTableAction.configureManagerComboBox(),
If previously selected memo is null, proxy default is now selected as initial selected option.
Fixes #9003
commit 581a7fbb (2021-03-14): Update create new Bean to prevent NPE

In AbstractTableAction.configureManagerComboBox(),
If previously selected memo is null, proxy default selected by default.
Fixes #9003
closed 2021-03-14 by danielb987