JMRI/JMRI #9007: Table checkboxes not initialized right

bobjacobsen, 2020-09-08:
If you start a LocoNet-type connection and look at the sensor table, you don’t see a pull-up column:

[image]

You do see a checkbox that will show it (see arrow).  When you check that box, a column appears:

[image]

Interestingly, the debounce information is showing in both cases, even though the box is not checked (red? arrow). If I check the box and then uncheck it, those columns disappear; they stay gone after restart.

From this I conclude there’s a confusing bug:  If one of the boxes is checked at shutdown, that state is remembered (OK) and controls the display next time (OK), but the box is not checked next time (Not OK, as it’s confusing)

That same bug seems to be present for Show Pull-up/Down Information and Show State Query actions checkboxes.

jerryg2003, 2020-09-08:
I found the checkboxes and the column selector pull down menu are not always consistent because selecting from the column selector does not update the checkboxes .  I can select columns in the pull down and it won’t affect the corresponding checkbox.  If I then click the checkbox twice (to check it and then uncheck it), it will deselect the appropriate columns.  My suggestion is to remove the columns from the column selector if they can be controlled from checkboxes (or vice versa if you prefer).  Only need one mechanism for controlling display of each column.

At the same time, the header names “State” on Query and Forget button columns could be removed re comments in issue #5491.

bobjacobsen, 2020-09-08:
Not sure what you mean by column selector?

jerryg2003, 2020-09-08:
"Column selector pulldown" is what I have been calling the list of columns you see when you right click on the column header.  Is there another name I should use in the Help?

KenC57, 2020-09-08:
Column selector: right clicking on a column header gives a checklist of all
the column names for displaying or hiding them.

PR #9410 (2021-02-19): Beantables - Column / Checkbox Improvements
Fix Light Table correct number columns (+1).
Add connection specific columns to Light Table. Fixes #7035
Do NOT display conn. specific Sensor columns in other Sensor tables.
Column select menu stays open after items checked / unchecked.
Conn. specific bottom checkbox updates on column visibility.
Sensor / Turnout checkboxes update on column visibility - fixes #9007 
Tri-State JCheckbox added - can indicate partially selected.
Eg. with Debounce column group partially visible, state query columns partially visible.
Nb, user can still only select / dese [... cut]
closed 2021-02-19 by danielb987