JMRI/JMRI #9023: Test Failures - Could not initialize plugin: interface org.mockito.plugins.MockMaker

icklesteve, 2020-09-17:
In latest Jenkins build #390 https://builds.jmri.org/jenkins/job/development/job/builds/390/ there are 597 new test failures.

Reproducing locally with 500+ failures with a whole jmrix test run, ie ant test-single -Dtest.includes=jmri.jmrix.*
which makes dev. a tad awkward.

Mockito was updated in #9009 , commit https://github.com/JMRI/JMRI/commit/d452f063ee11ce40cbcbe80455623c67f95c39b9
Although ant was updated at the same time, I'm reproducing this locally on Win7 standard JRE + org.eclipse.jdt.core.JDTCompilerAdapter JRE with the older ant version 1.10.7.

e.g. output for DebugThrottleTest testGetF12. 
https://builds.jmri.org/jenkins/job/development/job/builds/390/testReport/junit/jmri.jmrix.debugthrottle/DebugThrottleTest/testGetF12__/

```
Error Message
Could not initialize plugin: interface org.mockito.plugins.MockMaker (alternate: null)
Stacktrace
java.lang.IllegalStateException: Could not initialize plugin: interface org.mockito.plugins.MockMaker (alternate: null)
	at org.mockito.internal.configuration.plugins.PluginLoader$1.invoke(PluginLoader.java:82)
	at com.sun.proxy.$Proxy49.isTypeMockable(Unknown Source)
	at org.mockito.internal.util.MockUtil.typeMockabilityOf(MockUtil.java:29)
	at org.mockito.internal.util.MockCreationValidator.validateType(MockCreationValidator.java:22)
	at org.mockito.internal.creation.MockSettingsImpl.validatedSettings(MockSettingsImpl.java:250)
	at org.mockito.internal.creation.MockSettingsImpl.build(MockSettingsImpl.java:232)
	at org.mockito.internal.MockitoCore.mock(MockitoCore.java:68)
	at org.mockito.Mockito.mock(Mockito.java:1933)
	at org.mockito.Mockito.mock(Mockito.java:1844)
	at jmri.jmrix.debugthrottle.DebugThrottleTest.setUp(DebugThrottleTest.java:214)
	at sun.reflect.GeneratedMethodAccessor137.invoke(Unknown Source)
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.lang.reflect.Method.invoke(Method.java:498)
	at org.junit.platform.commons.util.ReflectionUtils.invokeMethod(ReflectionUtils.java:686)
	at org.junit.jupiter.engine.execution.MethodInvocation.proceed(MethodInvocation.java:60)
	at org.junit.jupiter.engine.execution.InvocationInterceptorChain$ValidatingInvocation.proceed(InvocationInterceptorChain.java:131)
	at org.junit.jupiter.engine.extension.TimeoutExtension.intercept(TimeoutExtension.java:149)
	at org.junit.jupiter.engine.extension.TimeoutExtension.interceptLifecycleMethod(TimeoutExtension.java:126)
	at org.junit.jupiter.engine.extension.TimeoutExtension.interceptBeforeEachMethod(TimeoutExtension.java:76)
	at org.junit.jupiter.engine.execution.ExecutableInvoker$ReflectiveInterceptorCall.lambda$ofVoidMethod$0(ExecutableInvoker.java:115)
	at org.junit.jupiter.engine.execution.ExecutableInvoker.lambda$invoke$0(ExecutableInvoker.java:105)
	at org.junit.jupiter.engine.execution.InvocationInterceptorChain$InterceptedInvocation.proceed(InvocationInterceptorChain.java:106)
	at org.junit.jupiter.engine.execution.InvocationInterceptorChain.proceed(InvocationInterceptorChain.java:64)
	at org.junit.jupiter.engine.execution.InvocationInterceptorChain.chainAndInvoke(InvocationInterceptorChain.java:45)
	at org.junit.jupiter.engine.execution.InvocationInterceptorChain.invoke(InvocationInterceptorChain.java:37)
	at org.junit.jupiter.engine.execution.ExecutableInvoker.invoke(ExecutableInvoker.java:104)
	at org.junit.jupiter.engine.execution.ExecutableInvoker.invoke(ExecutableInvoker.java:98)
	at org.junit.jupiter.engine.descriptor.ClassBasedTestDescriptor.invokeMethodInExtensionContext(ClassBasedTestDescriptor.java:481)
	at org.junit.jupiter.engine.descriptor.ClassBasedTestDescriptor.lambda$synthesizeBeforeEachMethodAdapter$18(ClassBasedTestDescriptor.java:466)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.lambda$invokeBeforeEachMethods$2(TestMethodTestDescriptor.java:162)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.lambda$invokeBeforeMethodsOrCallbacksUntilExceptionOccurs$5(TestMethodTestDescriptor.java:190)
	at org.junit.platform.engine.support.hierarchical.ThrowableCollector.execute(ThrowableCollector.java:73)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.invokeBeforeMethodsOrCallbacksUntilExceptionOccurs(TestMethodTestDescriptor.java:190)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.invokeBeforeEachMethods(TestMethodTestDescriptor.java:159)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.execute(TestMethodTestDescriptor.java:133)
	at org.junit.jupiter.engine.descriptor.TestMethodTestDescriptor.execute(TestMethodTestDescriptor.java:71)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$5(NodeTestTask.java:135)
	at org.junit.platform.engine.support.hierarchical.ThrowableCollector.execute(ThrowableCollector.java:73)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$7(NodeTestTask.java:125)
	at org.junit.platform.engine.support.hierarchical.Node.around(Node.java:135)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$8(NodeTestTask.java:123)
	at org.junit.platform.engine.support.hierarchical.ThrowableCollector.execute(ThrowableCollector.java:73)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.executeRecursively(NodeTestTask.java:122)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.execute(NodeTestTask.java:80)
	at java.util.ArrayList.forEach(ArrayList.java:1257)
	at org.junit.platform.engine.support.hierarchical.SameThreadHierarchicalTestExecutorService.invokeAll(SameThreadHierarchicalTestExecutorService.java:38)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$5(NodeTestTask.java:139)
	at org.junit.platform.engine.support.hierarchical.ThrowableCollector.execute(ThrowableCollector.java:73)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$7(NodeTestTask.java:125)
	at org.junit.platform.engine.support.hierarchical.Node.around(Node.java:135)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.lambda$executeRecursively$8(NodeTestTask.java:123)
	at org.junit.platform.engine.support.hierarchical.ThrowableCollector.execute(ThrowableCollector.java:73)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.executeRecursively(NodeTestTask.java:122)
	at org.junit.platform.engine.support.hierarchical.NodeTestTask.execute(NodeTestTask.java:80)
[... 49 lines cut ...]
JVM vendor version : 25.181-b13
JVM name           : Java HotSpot(TM) 64-Bit Server VM
JVM version        : 1.8.0_181-b13
JVM info           : mixed mode
OS name            : Linux
OS version         : 4.15.0-74-generic

	at org.mockito.internal.creation.bytebuddy.InlineByteBuddyMockMaker.<init>(InlineByteBuddyMockMaker.java:195)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at java.lang.Class.newInstance(Class.java:442)
	at org.mockito.internal.configuration.plugins.PluginInitializer.loadImpl(PluginInitializer.java:50)
	... 84 more
Caused by: java.lang.NoClassDefFoundError: net/bytebuddy/agent/ByteBuddyAgent
	at org.mockito.internal.creation.bytebuddy.InlineByteBuddyMockMaker.<clinit>(InlineByteBuddyMockMaker.java:108)
	... 90 more
Caused by: java.lang.ClassNotFoundException: net.bytebuddy.agent.ByteBuddyAgent
	at java.net.URLClassLoader.findClass(URLClassLoader.java:381)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:424)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:349)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:357)
	... 91 more
```

icklesteve, 2020-09-17:
This is affecting the ant builds.
When I do a clean build in NetBeans ( using Maven ), the tests run ok.

bobjacobsen, 2020-09-17:
I can’t recreate on my Mac with ant.  Specifically, this line works as expected:

% ant realclean tests && ./runtest.csh jmri.jmrix.debugthrottle.DebugThrottleTest

Could you post the commands you’re using please?

Bob

> 
> 
> This is affecting the ant builds.
> When I do a clean build in NetBeans ( using Maven ), the tests run ok.

icklesteve, 2020-09-18:
Hi Bob,

I've tried 
```
ant test-single -Dtest.includes=jmri.jmrix.*
ant clean test-single -Dtest.includes=jmri.jmrix.debugthrottle.DebugThrottleTest
ant realclean test-single -Dtest.includes=jmri.jmrix.debugthrottle.DebugThrottleTest
ant clean -Dbuild.compiler=org.eclipse.jdt.core.JDTCompilerAdapter test-single -Dtest.includes=jmri.jmrix.debugthrottle.DebugThrottleTest
```

JMRI ant was recently bumped to 1.10.8 , I was still using 1.10.7, now updated to 1.10.8, same issue.

The latest build and alltest runs ( ant 1.10.6 ? ) are also failing with this, so not just a windows thing.

https://builds.jmri.org/jenkins/job/development/job/builds/391/consoleFull
https://builds.jmri.org/jenkins/job/development/job/alltest/369/consoleFull

My next step in investigating this is to go back ( and possibly forwards ) with Mockito versions?

Steve.

bobjacobsen, 2020-09-18:
Looks like the problem is with the `test-single` target in build.xml. Maybe its paths were not updated properly.

icklesteve, 2020-09-18:
Closed by accident @Pugwash1 ?
Please re-open as is preventing the package builds.
Tia, Steve :-)

mattharris, 2020-09-18:
@icklesteve reopened

icklesteve, 2020-09-18:
Thanks @mattharris 

So . . .
I updated mockito to latest, change in error message, still failing.
I updated byte-buddy to latest, no change.
I then spotted that byte-buddy-agent was missing from build.xml and pom.xml , added and seems to be working.

Just running CI on own branch before submitting.

Pugwash1, 2020-09-18:
@icklesteve sorry

icklesteve, 2020-09-18:
No problem @Pugwash1 I've done it myself loads of times!

#9025 submitted.

closed 2020-09-20 by icklesteve