JMRI/JMRI #9029: Hold PRs for 24 hours

danielb987, 2020-09-19:
This is a follow up of https://github.com/JMRI/JMRI/pull/9027#issuecomment-695356068. I create a separate Issue since this is off topic for that PR.

In that comment, @bobjacobsen wrote:
> I tried to set up Mergeable to hold things for one day.  You can see some vestiges of it in the control files, but it didn’t work,  The basic problem is that GitHub actions are _action_ (event) based.  They don’t really have a concept of time that can be used for something like this.
> 
> The best workaround I could think of was to put a “Wait For It” label on the PR automatically. All that would be is a reminder, though, and it would be an extra step to remove it before merging.

Does JMRI has it's own server? If we have, one possibility is to have a cron script that runs maybe 6 times a day. That script could use [GitHub API](https://developer.github.com/v3/) to connect to GitHub, read the open PRs, check the creation date and time, and if 24 hours have passed, write a comment in the PR: `24 hours have passed. Can be merged`. The Mergable bot can then listen on comment messages in the PR, check the creation date of the PR and if 24 hours have passed, give green light to merge the PR.

The message "24 hours have passed. Can be merged" is only used to trigger a rerun of the Mergable so the message itself doesn't matter, except give users a clue of why it's there.

bobjacobsen, 2020-09-19:
A Jenkins job could run a script that does that. 

Such a script could be run from anywhere, so it could be developed locally and then moved to Jenkins.

mattharris, 2021-03-30:
Now implemented - see #9551

closed 2021-03-30 by mattharris