JMRI/JMRI #9030: alltest apps.gui3 test failures

icklesteve, 2020-09-20:
There are 2 unit tests failing the alltest build since 5th August, preventing the Jacoco builds.

https://builds.jmri.org/jenkins/job/development/job/alltest/311/

apps.gui3.mdi.MDITest.testCtor() - NPE in java.lang.NullPointerException
	at javax.swing.ImageIcon.<init>(ImageIcon.java:217)
	at jmri.util.swing.GuiUtilBase.actionFromNode(GuiUtilBase.java:44) . . .
apps.gui3.dp3.DecoderPro3Test.testCtor() - Unable to find frame "DecoderPro Wizard" to dispose.

I'm unable to reproduce these failures locally, can anyone else?

Steve

icklesteve, 2020-11-03:
jmri.jmrit.swing.meter.LoadAndStoreTest.loadAndStoreTest(File, boolean)[1] has also been failing since build 382
https://builds.jmri.org/jenkins/job/development/job/alltest/382/

Same failure in latest AllTest build attempt https://builds.jmri.org/jenkins/job/development/job/alltest/425/

icklesteve, 2020-11-07:
@danielb987 the changes in #9131 and #9132 didn't make any difference to AllTest failing.
Please re-open while I investigate further, thanks, Steve.

danielb987, 2020-11-07:
> @danielb987 the changes in #9131 and #9132 didn't make any difference to AllTest failing.
> Please re-open while I investigate further, thanks, Steve.

Done.

danielb987, 2020-11-07:
The reason that this Issue was closed was that I had linked the two PRs to this Issue. When the two PRs was merged, GitHub closed this Issue automatically.

If this happens again and you want to reopen the PR, feel free to do it.

icklesteve, 2021-03-13:
Looks like alltest started mainly passing again around build 462 / 463 :-)

closed 2021-03-13 by icklesteve