JMRI/JMRI #9031: PanelPro could no longer be started after upgrade to version 4.20

suremann99, 2020-09-20:
Hi JMRI Team
After upgrading JMRi to version 4.20, I could no longer start PanelPro. Get the following error message in the message.log file:

2020-09-17 15:47:32,270 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]
2020-09-17 15:47:32,285 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\admin\JMRI\log\messages.log [main]
2020-09-17 15:47:32,285 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\admin\JMRI\log\session.log [main]
2020-09-17 15:47:32,301 apps.Apps                             INFO  - PanelPro version 4.20+Rc7ba8249b starts under Java 1.8.0_261 on Windows 10 x86 v10.0 at Thu Sep 17 15:47:32 CEST 2020 [main]
2020-09-17 15:47:36,966 apps.Apps                             INFO  - Starting with profile DecoderPro.3f5972f6 [main]
2020-09-17 15:47:37,279 node.NodeIdentity                     INFO  - Using eee42e59-8947-4694-a121-a9c7e8b76f99 as the JMRI storage identity for profile id 3f5972f6 [AWT-EventQueue-0]
2020-09-17 15:47:37,404 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]
2020-09-17 15:47:37,841 pr3.PR3Adapter                        INFO  - PR3 adapter set hardware flow control, mode=2 RTSCTS_OUT=2 RTSCTS_IN=1 [main]
2020-09-17 15:47:37,841 locobuffer.LocoBufferAdapter          INFO  - COM6 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]
2020-09-17 15:47:37,903 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\admin\JMRI\ [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\admin\JMRI\DecoderPro\ [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\admin\JMRI\ [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path home: is C:\Users\admin\ [main]
2020-09-17 15:47:38,607 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]
2020-09-17 15:47:39,044 util.ColorUtil                        ERROR - unknown color text 'Black' sent to stringToColor [main]
2020-09-17 15:47:39,044 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]
java.lang.IllegalArgumentException: unknown color text 'Black'
	at jmri.util.ColorUtil.stringToColor(ColorUtil.java:189)
	at jmri.jmrit.operations.setup.Setup.setDropTextColor(Setup.java:1492)
	at jmri.jmrit.operations.setup.Setup.load(Setup.java:2432)
	at jmri.jmrit.operations.setup.OperationsSetupXml.readFile(OperationsSetupXml.java:77)
	at jmri.jmrit.operations.OperationsXml.load(OperationsXml.java:41)
	at jmri.jmrit.operations.setup.OperationsSetupXml.initialize(OperationsSetupXml.java:104)
	at jmri.InstanceManager.getInstance(InstanceManager.java:293)
	at jmri.InstanceManager.getNullableDefault(InstanceManager.java:233)
	at jmri.InstanceManager.getDefault(InstanceManager.java:204)
	at jmri.jmrit.operations.setup.Setup.isMainMenuEnabled(Setup.java:338)
	at apps.Apps.createMenus(Apps.java:497)
	at apps.Apps.createFrame(Apps.java:1065)
	at apps.PanelPro.PanelPro.main(PanelPro.java:122)

DecoderPro works well and can be started without any problems.

Many thanks in advance for any help with that issue.

Regards,

Markus

dsand47, 2020-09-20:
What version of JMRI were you previously using?

The problem is in the Operations.xml file.  The word Black needs to be lowercase.  There may be other incorrect color names.

suremann99, 2020-09-21:
Hi Dave

 

Thanks a lot for your fast reply!!

 

I’ve checked the old logs, and found out, that I still have had JMRI 4.8 running as I’ve done the upgrade to 4.20 :(

I saw too late in the release notes, that you should not do an direct upgrade to 4.20, if you’ve an older version of JMRI running.

Would it be possible to do a downgrade to 4.8 again or is this not advisable?

 

I’ve edited the Operations.xml (c:\Program Files (x86)/JMRI/xml/demoOperations) file and changed all the colors (Red, Blue, Black) to (red, blue, black) but I still can’t start PanelPro. Same Java exception as before. As it looks like, there is another xml file. I could only find the Operations.xml file at the previous mentioned location.

 

2020-09-21 11:49:59,914 util.Log4JUtil                        INFO  - ****** JMRI log ******* [main]

2020-09-21 11:50:00,362 util.Log4JUtil                        INFO  - This log is appended to file: C:\Users\admin\JMRI\log\messages.log [main]

2020-09-21 11:50:00,377 util.Log4JUtil                        INFO  - This log is stored in file: C:\Users\admin\JMRI\log\session.log [main]

2020-09-21 11:50:00,685 apps.Apps                             INFO  - PanelPro version 4.20+Rc7ba8249b starts under Java 1.8.0_261 on Windows 10 x86 v10.0 at Mon Sep 21 11:50:00 CEST 2020 [main]

2020-09-21 11:50:10,347 apps.Apps                             INFO  - Starting with profile PanelPro.3deb2b69 [main]

2020-09-21 11:50:11,873 node.NodeIdentity                     INFO  - Using eee42e59-8947-4694-a121-a9c7e8b76f99 as the JMRI storage identity for profile id 3deb2b69 [AWT-EventQueue-0]

2020-09-21 11:50:12,591 xml.AbstractSerialConnectionConfigXml INFO  - Starting to connect for "LocoNet" [main]

2020-09-21 11:50:12,591 jmrix.AbstractSerialPortController    WARN  - old profile format port speed value converted [main]

2020-09-21 11:50:14,837 pr3.PR3Adapter                        INFO  - PR3 adapter set hardware flow control, mode=2 RTSCTS_OUT=2 RTSCTS_IN=1 [main]

2020-09-21 11:50:14,900 locobuffer.LocoBufferAdapter          INFO  - COM6 port opened at 57600 baud with DTR: true RTS: true DSR: false CTS: false  CD: false [main]

2020-09-21 11:50:15,501 loconet.LnPacketizer                  INFO  - lnPacketizer Started [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path program: is C:\Program Files (x86)\JMRI\ [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path preference: is C:\Users\admin\JMRI\ [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path profile: is C:\Users\admin\JMRI\PanelPro\ [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path settings: is C:\Users\admin\JMRI\ [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path home: is C:\Users\admin\ [main]

2020-09-21 11:50:20,116 util.FileUtilSupport                  INFO  - File path scripts: is C:\Program Files (x86)\JMRI\jython\ [main]

2020-09-21 11:50:22,825 util.ColorUtil                        ERROR - unknown color text 'Black' sent to stringToColor [main]

2020-09-21 11:50:22,825 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [main]

java.lang.IllegalArgumentException: unknown color text 'Black'

                at jmri.util.ColorUtil.stringToColor(ColorUtil.java:189)

                at jmri.jmrit.operations.setup.Setup.setDropTextColor(Setup.java:1492)

                at jmri.jmrit.operations.setup.Setup.load(Setup.java:2432)

                at jmri.jmrit.operations.setup.OperationsSetupXml.readFile(OperationsSetupXml.java:77)

                at jmri.jmrit.operations.OperationsXml.load(OperationsXml.java:41)

                at jmri.jmrit.operations.setup.OperationsSetupXml.initialize(OperationsSetupXml.java:104)

                at jmri.InstanceManager.getInstance(InstanceManager.java:293)

                at jmri.InstanceManager.getNullableDefault(InstanceManager.java:233)

                at jmri.InstanceManager.getDefault(InstanceManager.java:204)

                at jmri.jmrit.operations.setup.Setup.isMainMenuEnabled(Setup.java:338)

                at apps.Apps.createMenus(Apps.java:497)

                at apps.Apps.createFrame(Apps.java:1065)

                at apps.PanelPro.PanelPro.main(PanelPro.java:122)

 

Regards,

 

Markus

 

This email has been checked for viruses by AVG.
https://www.avg.com

mattharris, 2020-09-21:
Markus,

Which locale are you running JMRI under? Only there are some locale-specific parts for the colour name determination.

In my quick test, using en-GB, having colour 'Black' is fine. However, if you are running in a non-English locale, the locale-specific values _may_ be being used instead.

Also, the files you probably need to be looking at are in your user preferences location - from the log, you should find them under 'C:\Users\admin\JMRI\PanelPro\' - this is where you need to search out the 'Operations.xml' and not the one in the JMRI installation directory.

Hope this helps.

Best regards,

Matt H

mattharris, 2020-09-21:
Markus,

A bit more testing, and the only line that appears to need changing in `Operations.xml` to permit a load is:

`<manifestColors dropColor="red" pickupColor="blue" localColor="black" />`

All the colours for `manifestColors` need to be in the lower-case form.

This is based upon an attempted load of the Operations Demo files whilst running under the nl-BE locale.

Best regards,

Matt H

KenC57, 2020-09-21:
If you had JMRI running while trying to do an upgrade, you get a mixed bag
of parts due to some of the files being locked while in use. I would say,
since you had been at 4.8, I suggest the following:

1. Save the current contents of the \Program Files (x86)\JMRI by copying it
to a new name like JMRI_Old. The idea is to preserve any modifications you
might have put in that file tree (instead of the user space where they would
be safe between updates).
2. Next is do a uninstall JMRI via the control panel tool in Windows for
removing programs.
3. Look to see if there is any leftover parts in the \Program Files
(x86)\JMRI, remove that too by deleting.
4. Now you have a clean space to start again, program wise.
5. Download and install 4.12. Try starting your profile. If it seems ok, do
as it instructs in the release notes, save the preferences and save the
panel file (to a new name) and exit.
6. Download and install 4.18. Do the same set of steps as you did with 4.12.
Same thing, save preferences and panel.
7. Download and install 4.20. Test and see how it goes.

If you do a step and something doesn't seem right, ask for more help.

-ken c

suremann99, 2020-09-21:
Hi Matt, Dave  and Ken

 

 

You guys are the best!!!

Problem solved!! PanelPro is working again!!

 

I edited the c:\users\admin\JMRI\Operations\Operations.xml file from :

 

<manifestColors dropColor="Black" pickupColor="Black" localColor="Black" />

 

to

 

<manifestColors dropColor="black" pickupColor="black" localColor="black" />

 

Now PanelPro is starting again and I can operate my layout.

 

Many many thanks to all of you for that fast solution finding!! Highly appreciated!!

 

Regards,

 

Markus

 

https://www.avg.com

closed 2021-04-14 by stale[bot]