JMRI/JMRI #9055: Provide default units, range, and precision for Meters

devel-bobm, 2020-10-01:
Meters rely upon GUI settings for "range and precision" (i.e. # integer digits and # decimal digits), and units.  It may be better for the concrete Meter implementation to provide suggested values to the GUI, and allow the user to override those suggestions. 

In a review comment to P/R #9050, @danielb957 suggested"

One simple solution for this is to add this to the Meter interface:

```
/**
 * Returns the number of integer digits this meter recommends for the display.
 *
 * @returns the number of integer digits, or 0 if no recommendation is given
 */
public default int getRecommendedIntegerDigits() {
    return 0;
}

/**
 * Returns the number of decimal digits this meter recommends for the display.
 *
 * @returns the number of decimal digits, or 0 if no recommendation is given
 */
public default int getRecommendedDecimalDigits() {
    return 0;
}
```

The MeterFrame would when check `Meter.getRecommendedIntegerDigits()` to check if it's value is non zero and if so, use it. The LnPredefinedMeters would then override these methods to return the desired number of digits. This would not break any of the other meters so it would be safe to add.

_Originally posted by @danielb987 in https://github.com/JMRI/JMRI/pull/9050#r497782192_

devel-bobm, 2020-10-02:
@danielb987 , I found an alternate way to at least partially address this is for `MeterFrame.setMeter()` to use the `Meter.getMax()` and `Meter.getResolution()` methods (inherited from `AnalogIO`) and figure out an optimal number of digits from there.  When the meter is actually selected, this overrides any previously-set GUI options, so that the meter is displayed based on computed digits numbers.

There is always a difficult decision between object-implied settings and GUI settings.  I chose to keep the object-implied settings and ignore the GUI settings, knowing that a user who really wanted to display a lesser or greater number of digits, whether significant or not, could always do so.  

I notice that if the user sets a smal number of digits to the left of the decimal point, and the value being displayed requires more digits than the user configures, then the display _will_ display the required number of digits, rather than truncating the value or displaying any "out-of-range" error message.  Based on this behavior, I consider ignoring the user settings on meter selection to be the better choice than blindly honoring the user settings, regardless of the object's capabilities.

danielb987, 2020-10-02:
> I found an alternate way to at least partially address this is for `MeterFrame.setMeter()` to use the `Meter.getMax()` and `Meter.getResolution()` methods (inherited from `AnalogIO`) and figure out an optimal number of digits from there.

It works fine for the default, but it may not be what the user wants.

Example: A meter measures track current and is able to measure 0.00 - 3.00 ampere. But the user has a N-scale layout that draws less than 200 mA. Using getMax() and getResolution() will result in a meter that shows 0.00 - 3.00 ampere, but the user of this N-scale layout want a display that shows 0 - 200 mA.

> I notice that if the user sets a smal number of digits to the left of the decimal point, and the value being displayed requires more digits than the user configures, then the display _will_ display [...]

If the display wants to show 2 digits, but the number often gets three numbers, the flicker between two and three digits may be irritating. Think of a meter there this happens once every second. If the measurement goes between 95 mA and 105 mA, this can happen very often and therefore be very irritating for the user, since changing the number of digits to be displayed will change the size of the digits. For LocoNet, this is not a problem since the update intervall is slow, but for other connections this can be a real problem.

bobjacobsen, 2020-10-02:
> I chose to keep the object-implied settings and ignore the GUI settings, knowing that a user who really wanted to display a lesser or greater number of digits, whether significant or not, could always do so.

I don’t understand this.  If the GUI settings are being ignored, how does the user change the number of digits?

The usual hierarchy is:
 - If there are user settings, they win
 - If not and there are hardware/system specific settings, they win
 - Otherwise the general defaults are used.

Perhaps I misunderstood what “GUI settings” means?

devel-bobm, 2020-10-02:
@bobjacobsen , 

"GUI settings" are selections in a menu pull-down:
[image]

Let me re-phrase the statement you are commenting upon:  When accepting the selection of some `Meter` object, my proposed solution chooses to ignore the "Settings".  The proposed implementation does _not_ limit a user's ability to change the display settings at any later time, or limit the user's ability to choose Settings values which do not make sense for the actively-displayed `Meter` object..

If there is a desire that GUI settings "always win", even if they are non-sensical, then I would suggest that the GUI would be updated to provide some "clues" to users when the settings don't make sense.

- Perhaps "Out-of-range" when the value is greater than the largest displayable value;

- Perhaps some sort of down-arrow when the displayed value is 0 but there are valid non-zero digits available.

And I would suggest that some way be provided to allow the `Meter` object to provide "default" settings, such as when there are no pre-selected (or pre-loaded) GUI "Settings".  This implies that the code should have the ability to "persist" a Meter GUI _with_ persistence of "unselected" number of integer digits and/or "unselected" number of decimal digits and/or "unselected" "Units" value.

devel-bobm, 2020-10-02:
@danielb987 wrote:

> If the display wants to show 2 digits, but the number often gets three numbers, the flicker between two and three digits may be irritating. Think of a meter there this happens once every second. If [...]

This is a real problem with the existing code.

bobjacobsen, 2020-10-02:
@devel-bobm You didn't say _HoW_ the GUI settings could sometimes be used (when the user wants them to  be), and sometimes not, which was my question based on what you said in your earlier comment.

You're concerned about non-sensical user settings.  I'm concerned about undesirable hardware defaults, for example the cases previously mentioned by @danielb987.  Perhaps there's way to prevent both issues:

How about:
 - If a box is checked in a section of the Settings menu, that value is used
 - If none of the boxes in a section of the Settings menu are checked, the value provided by the underlying hardware-specific option will be used
 - There are default methods to provide generic values for each section if a hardware-specific implementation doesn't override them

danielb987, 2020-10-02:
> Let me re-phrase the statement you are commenting upon: When accepting the selection of some `Meter` object, my proposed solution chooses to ignore the "Settings". The proposed implementation does [...]

That's fine for me.

One thing that I think needs to be added to the MeterFrame is that if the MeterFrame is loaded from a panel file and its meter doesn't exist yet, the MeterFrame should aquire the meter once it's created. This can be done by adding the field "systemName" to MeterFrame and let MeterFrameManagerXml set MeterFrame.systemName when the MeterFrame is loaded by MeterFrameManagerXml.

bobjacobsen, 2020-10-03:
It’s to fine for me.  It’s so massively _not_ _fine_ for me that I must be not understanding it correctly.

What’s the point of even having a Settings menu when a "proposed solution chooses to ignore the "Settings”.”?

danielb987, 2020-10-03:
@bobjacobsen 
I think you missunderstand @devel-bobm intentions. I did too.

I think we all three wants the same thing, but we express it different which leads to misunderstanding.

bobjacobsen, 2020-10-03:
I'm quite sure I don't understand it. I've now said that several times.  But I would very much like @devel-bobm to explain how his idea works, not in the case he's repeatedly mentioned, but in the one I'm asking about:  _How_, exactly, is this supposed to work such that the user can override the initial settings from the hardware and have that those selections work at that time and in the future?

Pugwash1, 2020-10-03:
The usual hierarchy is:
 - If there are user settings, they win
 - If not and there are hardware/system specific settings, they win
 - Otherwise the general defaults are used.

It may be better for the concrete Meter implementation to provide suggested values to the GUI, and allow the user to override those suggestions.

These 2 statements appear to me to pretty much the same thing.

The assumption in both implies users can make  changes and settings are saved and those are used next time.

KenC57, 2020-10-03:
My thought for the options would be the user preference records keep
whatever you change it to. It would come up first time with a default of
settings. You change and close the window, that saves the settings, like
many other windows (like the connection monitors). Then the only issue is
preventing some incompatible set of settings that would prevent at least
some form of the window from being displayed.

danielb987, 2020-10-03:
I think it's better to save the meter frames in the panel file as it is today. The reason is that different panel files may have different connection types with different sets of meters.

The way I think MeterFrame should work is the following:

* When the user selects a new meter in the MeterFrame, the MeterFrame tries to get number of digits from the meter. For example by the Meter.getMax() and Meter.getResolution() methods inherited from AnalogIO.
* When a panel is loaded that contains MeterFrames, the MeterFrames that are restored tries to open its meter and reads the number of digits from the panel file.
* If the meter doesn't exist when the panel is loaded, for example if the panel is loaded at startup and JMRI hasn't yet discovered the LocoNet meters, the MeterFrame should wait for the meter to be registered. Once the meter is registered in the manager, the MeterFrame automatically connects to it. But keep the number of digits as it was stored in the panel file.

I never implemented this in PR #9033 since in that PR all the existing meters where created before any panel is loaded.

devel-bobm, 2020-10-04:
@bobjacobsen:

I will describe how my proposed solution behaves for a few cases:

Case 1: Tool is opened manually before any Meter is known, no Settings modification by user:
1a) User opens Voltage/Current Meter tool
1b) User _cannot_ select a meter, because none are "known" yet.
1c) User _can_ configure in Settings menu "# integer digits" and "# decimal digits", but user does not do so.  (User is not offered "units" settings in current implementation for reasons unknown to me!)
1d) A LocoNet-based Voltage Meter object is created with "max" value of 25.4 and resolution of 0.2V.  Effectively simultaneously, a LocoNet-based Current Meter object is created with "max" value of 12.7 and resolution of 0.1A.  
1e) Both become known to the GUI effectively simultaneously, and they populate to the Voltage Meters pull-down menu and Current Meters pull-down menu, as appropriate.
1f) Some time later, the user selects the populated Voltage meter.
1g) My proposed solution extracts the max value and resolution from the appropriate Meter object's fields.  It computes that the max value of 25.4 requires 2 "integer digits" and computes that the resolution implies 1 "decimal digit".  It uses these values to display the current voltage reading from the Voltage Meter object.
1h) Because the GUI has a selected meter, the "Update" mechanism is enabled, and periodically, that mechanism triggers a LocoNet query.
1i) As LocoNet messages are seen which update the Meter objects, the GUI sees property change events from the selected Meter object, and updates the display based on the current value and the value of the Setttings at that time.
1j) If the user changes any of the "Settings" values - # integer digits, #decimal digits, Units (note that the Units checkboxes appear in the menu _only_ after selection of a Meter!), the display attempts to use them from that point forward.  If the # integer digits is less than the minimum # integer digits needed to display the value, the display will show all of the integer digits, regardless of the setting (but does _not_ change the GUI setting!) - i.e. if # integer digits is 1 and value is 1502, the GUI will display all 4 digits, rather than just the one that is specified in Settings for "# integer digits".  If the user has specified a "# decimal digits", the GUI will display _only_ those digits, by truncating all digits of lesser-significance.  If the user changes the "units", that appropriate multiplier will be applied to the value before the GUI displays it - that value resulting from the multiplication will be displayed following the guidlines  listed above for "integer digits" and "decimal digits".
1k) If the user changes any of the settings again, all of the comments from 1j) above apply.

Case 2: Tool is opened manually before any Meter is known, Settings are modified by user before Meter selection occurs:
2a) User opens Voltage/Current Meter tool
2b) User _cannot_ select a meter, because none are "known" yet.
2c) User _can_ configure in Settings menu "# integer digits" and "# decimal digits", and selects 4 integer digits and 3 decimal digits.  (User is not offered "units" settings in current implementation for reasons unknown to me!)
2d) A LocoNet-based Voltage Meter object is created with "max" value of 25.4 and resolution of 0.2V.  Effectively simultaneously, a LocoNet-based Current Meter object is created with "max" value of 12.7 and resolution of 0.1A.  
2e) Both become known to the GUI effectively simultaneously, and they populate to the Voltage Meters pull-down menu and Current Meters pull-down menu, as appropriate.
2f) Some time later, the user selects the populated Voltage meter.
2g) My proposed solution extracts the max value and resolution from the appropriate Meter object's fields.  It computes that the max value of 25.4 requires 2 "integer digits" and computes that the resolution implies 1 "decimal digit".  It uses these values to display the current voltage reading from the Voltage Meter object.  The settings applied in 2c) are _ignored_.
2h) Because the GUI has a selected meter, the "Update" mechanism is enabled, and periodically, that mechanism triggers a LocoNet query.
2i) As LocoNet messages are seen which update the Meter objects, the GUI sees property change events from the selected Meter object, and updates the display based on the current value and the value of the Setttings at that time.
2j) If the user changes any of the "Settings" values - # integer digits, #decimal digits, Units (note that the Units checkboxes appear in the menu _only_ after selection of a Meter!), the display attempts to use them from that point forward.  If the # integer digits is less than the minimum # integer digits needed to display the value, the display will show all of the integer digits, regardless of the setting (but does _not_ change the GUI setting!) - i.e. if # integer digits is 1 and value is 1502, the GUI will display all 4 digits, rather than just the one that is specified in Settings for "# integer digits".  If the user has specified a "# decimal digits", the GUI will display _only_ those digits, by truncating all digits of lesser-significance.  If the user changes the "units", that appropriate multiplier will be applied to the value before the GUI displays it - that value resulting from the multiplication will be displayed following the guidlines  listed above for "integer digits" and "decimal digits".
[... cut ...]
5f) User changes Settings (# integer digits, # decimal digits, Units) as desired.  Display updates as described in 1j).

I recognize that the behaviors described above are _not_ what you desire.  I can envision ways to attempt to meet the behaviors you have previously mentioned, in ways that give the behaviors you desire in all of the cases defined above.  The one exception I can think of would be that I do not at this time have a good way in mind for case 5e to use persisted settings from 5b).

danielb987, 2020-10-04:
> Case 3: Tool is opened via a panel XML file before any Meter is known, Settings are modified by user before Meter selection occurs: 3a) Panel file (with saved Meter GUI) is loaded. My current code gives NPE in MeterFrameManagerXML.java account no available meter with the given UUID.

This can easily be solved if the MeterFrame has the system name of the meter. When the MeterFrame is loaded, it tries to find the meter with the desired system name. If no meter with that system name exists, it stores the system name and wait for that system name to reappear. When that meter is created, the MeterFrame starts to use it.

The selected unit is also stored in the panel file so if the MeterFrame saves that setting, it can restore the selected unit as well when the meter is created.

danielb987, 2020-10-04:
> Case 4: Meter object is created _before_ the tool is opened via a panel XML, XML specifies an alread-created object. 4a) A LocoNet-based Voltage Meter object is created with "max" value of 25.4 and [...]

I think this is wrong. If the MeterFrame is stored in the panel file and then restored when the panel is loaded, JMRI must assume that it's the same meter that now exists that existed when the panel was stored.

Example:
A LocoNet meter is found and given the name LVXYZ1 with a measuring interval 0.00 - 30.00 volts. But the user uses this meter to measure voltage in the interval 0 - 200 mV. So the user selects 3 integer digits, 0 decimal digits and the unit mV. The panel is then stored.

JMRI is then restarted. The LocoNet meter LVXYZ1 with a measuring interval 0.00 - 30.00 volts is created.
When the panel is loaded, it loads the MeterFrame with the meter LVXYZ1. The meter suggest two integer digits, two decimal digits and the unit voltage, but the user still uses it to measure 0 -200 mV, so the user setting should have higher priority since the MeterFrame is loaded from a panel file.

devel-bobm, 2020-10-04:
@danielb987  wrote:
> This can easily be solved if the MeterFrame saves the system name of the meter. When the MeterFrame is loaded, it tries to find the meter with the desired system name. If no meter with that system name exists, it stores the system name and wait for that system name to reappear. When that meter is created, the MeterFrame starts to use it.

When I saved the profile, there was an active meter displayed, so the XML included:
```
  <meters class="jmri.managers.configurexml.AbstractMeterManagerXml">
    <meter>
      <systemName>LVDCS240(s/n216)InputCurrent</systemName>
    </meter>
    <meter>
      <systemName>LVDCS240(s/n216)Voltage</systemName>
    </meter>
  </meters>
```
And the NPE happened.  I have _not_ modified `MeterManagerConfigurationXML.java`.

@danielb987 also commented:
>I think this is wrong. If the MeterFrame is stored in the panel file and then restored when the panel is loaded, JMRI must assume that it's the same meter that now exists that existed when the panel was stored.

I am sorry that you think that the behavior I reported "is wrong".  I am just reporting the behavior I experienced with the code modified to behave as I have proposed.

Remember, the use-case behavior report I provided is _only_ a report of the behavior of the code _when it has the modifications which I have proposed_.  Whether those modifications are good or bad, or whether they can be part of any modifications that become an improvement the behavior of the GUI-portion and/or persistence portion of the `Meters` functionality.

danielb987, 2020-10-04:
I apologize.

When I made the split of MultiMeter into Meter and the work with MeterFrame, I did not add support for MeterFrame to listen on MeterManager for newly registered meters. I left that feature for later. You add that feature in #9050 which I thinks is great.

If you want, we can merge #9050 and I can fix the things that I mention in my two comments. That would be easy for me to do.

bobjacobsen, 2020-10-04:
I agree with @danielb987's points about restoring saved information.

Ideally, Meters would use the same hierarchy of preferences that the other JMRI tool windows do.  They're already storing their user-modified size and position via preferences. Adding the other settings would make them behave in expected ways.

It the concern is "Somebody might request fewer digits than needed, and that's Bad, even if they want it", you could do what most of my real meters do:  If they can't display the number, they display dashes or asterisks.  It's just another form of out-of-range.

devel-bobm, 2020-10-04:
I will not persue my proposed changes at this time.  I will consider this issue to be one of "ideas" for improvements to the functionality of the `MeterFrame` feature, and allow others to decide if/how to implement any of the ideas presented herein.

mstevetodd, 2021-03-30:
I was unaware of this discussion, and made changes to the default behavior of the Meters in #9314. Not sure if that solved this problem or made it worse.

closed 2021-04-14 by stale[bot]