JMRI/JMRI #9056: Splitval compound qualifier does not respond properly and it's decoder template throws an NPE on close

n3ix, 2020-10-02:
Attached is a simple decoder template with a Splitval defined as follows:
			<variable item="Qual" CV="11" mask="XXXVXXVX" default="0">
				<splitVal highCV="12" upperMask="XXXXVXXX" offset="0" /> 
			</variable>
This particular splitval mask yields possible values for Qual of {0,1,8,9,16,17,24,25}.  

The template has a pane called "QualTest" to make it easy to manipulate the value of Qual, along with three qualified panes: Qual != 0, Qual != 16, Qual != 0 and Qual != 16.   

The first two qualified panes work as expected.  The third pane does not work as expected.  Once the value of Qual is changed, the third pane ignores the first condition (Qual != 0) and reacts only to the second conditions (Qual != 16).

The attached template uses this form of compound qualifier that passes XML file validation:
        <qualifier>
            <variableref>Qual</variableref>
            <relation>ne</relation>
            <value>0</value>
        </qualifier>
        <qualifier>
            <variableref>Qual</variableref>
            <relation>ne</relation>
            <value>16</value>
        </qualifier>

When the template is closed, it throws the following:
2020-10-01 17:46:54,857 ptionhandler.UncaughtExceptionHandler ERROR - Uncaught Exception caught by jmri.util.exceptionhandler.UncaughtExceptionHandler [AWT-EventQueue-0]
java.lang.NullPointerException
	at jmri.jmrit.symbolicprog.SplitVariableValue.getLongValue(SplitVariableValue.java:507)
	at jmri.jmrit.symbolicprog.ArithmeticQualifier.currentDesiredState(ArithmeticQualifier.java:61)
	at jmri.jmrit.symbolicprog.QualifierCombiner.currentDesiredState(QualifierCombiner.java:37)
	at jmri.jmrit.symbolicprog.QualifierCombiner.propertyChange(QualifierCombiner.java:26)
	at java.beans.PropertyChangeSupport.fire(Unknown Source)
	at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
	at java.beans.PropertyChangeSupport.firePropertyChange(Unknown Source)
	at java.awt.Component.firePropertyChange(Unknown Source)
...
JMRI cannot be exited at this point and must be killed via the Task Manager.

I also tried an alternate form of compound qualifier as a test:
        <qualifier>
            <variableref>Qual</variableref>
            <relation>ne</relation>
            <value>0</value>
            <variableref>Qual</variableref>
            <relation>ne</relation>
            <value>16</value>
        </qualifier>

This second form of qualifier does not pass validation.  It only responds to the first condition and ignores the second, the opposite of the other form.  Interestingly the second form does not throw an NPE when the template is closed.

This is as far as I can figure out how to go on my own.  Hopefully there is enough here for someone to make sense of it.

[QualifierIssue.zip](https://github.com/JMRI/JMRI/files/5315433/QualifierIssue.zip)

n3ix, 2021-03-11:
Confirmed this is still present in 4.22.  Attached file demonstrates the calculation problem as well as throwing an exception when the template is closed.

dheap, 2021-03-12:
@n3ix I wonder if this is related to #9495?

n3ix, 2021-03-12:
Dave,

The qualifiers in the test decoder template only check for specific numeric values of a splitval. I looked at it but didn't see where "item" would come into it, so not sure that issue directly applies here?

Robin

Robin Becker

San Diego CA

dheap, 2021-03-12:
@n3ix
>The qualifiers in the test decoder template only check for specific numeric values of a splitval. I looked at it but didn't see where "item" would come into it, so not sure that issue directly applies here?

In every qualifier we have:
<variableref>someVariable</variableref>

In this 'someVariable' can match either:
... Item="someVariable" ...
or 
... label="someVariable" ...

We have realised match by 'label' can break down if the locale is not English. You need to match by 'item' for it to work in all cases.

Please check all XML where you have used qualifiers to make sure <variableref> Is always to the 'item' attribute and not the 'label' attribute.

dheap, 2021-03-12:
Sorry, markdown messed it up:

In every qualifier we have:
`<variableref>someVariable</variableref>`

n3ix, 2021-03-12:
@dheap  The variable used by all the qualifiers in the attached QualifierIssue demo template have item="Qual" as its first entry.  All the qualfiers themselves have <variableref>Qual</variableref>  So I think the template conforms?

n3ix, 2021-03-13:
It looks like the crash is due to qualifier property change listener not getting disconnected prior to the pane get disposed.     The result is that SplitVariableValue.getLongValue() is called with _textField=null.  

/**
     * Get the value as an unsigned long.
     *
     * @return the value as a long
     */
    @Override
    public long getLongValue() {
        return ((getValueFromText(_textField.getText()) - mOffset) / mFactor);
    }

n3ix, 2021-04-13:
Closed by 9577

closed 2021-04-13 by n3ix