JMRI/JMRI #9073: Occupancy Block - Signal Table display issue

jhitchens, 2020-10-12:
Hi folks,
I’m very new to JMRI, so apologies if this is not the right place to post this possible bug report.
I’m running PanelPro 4.20.
I’m having a problem with the Signal Table in the ‘Occupancy Blocks, their Portals and Paths’ window.
I start with an empty Signal Table:
 [image]

I add a signal with block/portal information:
 [image]

and press enter:
 [image]

The signal appears duplicated on 2 lines.
I ignore that, and add another signal:
[image]

and press enter:
 [image]

and that one is now shown on 2 lines.  If I go up and delete one of the lines or overwrite it, then the deletion or overwriting doesn’t happen to the signal that I hoped it would, and I end up losing or overwriting the wrong data.

The workaround is to ignore what the table shows you, and just keep typing your signals into the blank row.  Internally I think the data you enter is ok, but the table that is displayed somehow is mixed up, showing the last entry twice (even if sometimes they are not in adjacent rows in the displayed list).  In the end by ignoring the display and typing all my signals in, I managed to update the table.  I then saved and closed PanelPro and restarted it, and when I brought the table up again it looked ok.

Regards,
Jack

silverailscolo, 2020-10-23:
Notice that one of the duplicates shows length in cm (metric) and the other is in inches: [image]

jhitchens, 2020-10-23:
When I first started typing the blocks in, I was changing them from inches to cm, and I think cm became the default.  My guess is that the one with 'in' next to it is the 'ghost' entry.  But this bit was of least concern, unless it provides some clue as to what is happening internally.

silverailscolo, 2020-11-16:
Ghost entry fixed in PR #9142, thanks for reporting.
Please close the issue after you receive the next release and confirm this is solved.

closed 2020-12-07 by silverailscolo