JMRI/JMRI #9074: Failure to fully delete Portals

jhitchens, 2020-10-12:
Hi folks,
I’m running PanelPro 4.20.
I set up some Portals in the ‘Occupancy Blocks, their Portals and Paths’ window.
I discovered that I had duplicated some, so I deleted the ones that I didn’t want.
The deleted ones disappeared from the Portal Table.
However, when I ran the ‘Block-Portal Cross Reference’ tool I saw that my deleted portals were still in the system.
There were also some warnings in the Block/Portal/Path Warnings window (sorry, didn't record the exact message) about some of the deleted Portals not having connections.
Is it possible that the deletion isn’t clearing out the portals completely?
As a workaround I edited the .xml file directly, found the offending items and remove their <portal … </portal> entries, and the system is no longer complaining about them.  I hope this shouldn’t break anything else.  I did confirm that if I edited the .xml file incorrectly then the system would give me plenty of error messages :) so that gives me some level of confidence.
Regards,
Jack

silverailscolo, 2021-01-19:
I don't see such left-overs after deleting a complete Portal in 4.21.4ish, but instead am now looking into incomplete storing of OBlock changes to a Portal. (leads to error messages "no path into block...")

silverailscolo, 2021-01-31:
Not showing in 4.21.3. Additional issue fixed with PR #9339. Closing.

closed 2021-01-31 by silverailscolo